onlaj/Piano-LED-Visualizer #187: Problem with piano input

ababnoy, 2021-02-09:
https://user-images.githubusercontent.com/78776244/107314260-3848c600-6acf-11eb-9a0f-c665717f0075.mp4

I think my connections were properly connected but the lights don't respond to the input... I am using USB midi in/out btw, could that be the problem?

SivanDuijn, 2021-02-10:
Do the led animations work?

ababnoy, 2021-02-11:
No, it didn't work

ababnoy, 2021-02-11:
https://user-images.githubusercontent.com/78776244/107604170-1e8fb600-6c6a-11eb-9cb1-2dee1721168f.mp4

I got the LED animations working but only three LED's are functioning

SivanDuijn, 2021-02-13:
You probably have already tried this, but maybe turn the led count up to 176?

ababnoy, 2021-02-14:
Yes... and I also tried connecting the red wire to 5v but still nothing changed

ababnoy, 2021-02-17:
Ok, the La Campanella played which confirms that the RPI is receiving data from the piano... I think the LED strip has a bad sector.

ababnoy, 2021-03-04:
I bought a new LED strip which all the lights are now working on the animation but it is still now responding when I press them... I think the problem is withw the software

closed 2021-08-16 by onlaj