onlaj/Piano-LED-Visualizer #189: LED does not respond

MrNazers, 2021-02-09:
Hello, my LED strip does not work with the software, it works when I connect the data to port 14 of the gpio (RGB). The campanella doesn't work either, I have already tested the connections to the multimeter and the led band receives a 5.5V and 3.8V for the data.
My keyboard is well recognized by the raspberry, and the different led test doesn't work (neopixel error or no light).

[image]

SivanDuijn, 2021-02-10:
You could try to connect the data wire of the led strip to gpio pin 18 and test the led animations.
[image]

Hope this works

MrNazers, 2021-02-12:
Thanks Sivan , I checked all the connections, connected to gpio 18 ,then 12 as indicated in an old post,but the leds remain off :(

MrNazers, 2021-02-13:
I changed the raspberry, still the same problem I'm starting to think it's coming from my cabling. 

[image]

I have removed the red and white wire from the LED because they are useless, I'm not sure about the surrounded cabling.

SivanDuijn, 2021-02-13:
yes you don't those wires, but the other wires should be connected as follows:
[image]
So in you image this would look like
[image]

From the led strip you need to connect the white wire to the black ground cable coming from the power supply (and the pi), the green wire to the white signal wire coming from the pi, and the red wire to the red wire coming the from the power supply.

This should work

MrNazers, 2021-02-14:
thank you very much! At the beginning it didn't work but the problem came from one of the segment of the LED that wasn't working, so check it out, I finalized everything, it works great, but last problem since I removed a segment of the LED strip the keys are shifted ...you know how to tell the software to start at the 30th led for example ?

antonio-qp, 2021-02-14:
LED Strip Settings > Shift

MrNazers, 2021-02-18:
The shift works but it doesn't solve all my problem ^^ I tried to change the led count too but nothing to do the LED's are stuck at number 64. Any solutions ? 

[image]

closed 2021-02-24 by MrNazers