onlaj/Piano-LED-Visualizer #195: How to set color on a key by key basis with Sequences

mrdjnash, 2021-02-21:
I'm trying to set the color for each key on a key by key basis (I'm using code to generate the xml content). How do I structure the sequence setting to tell the system which key is which colour?

Thanks in advance.

EX8F22, 2021-08-10:
Mrdjnash, did you ever figure out how to do this? I also would like to change the color for each key on a key by key basis and do not know how.

mrdjnash, 2021-08-13:
Hi there,

Yes I did I think. You do it as a multicolor type and set the RGB color for
each key as a separate range. I've attached an XML file with each key from
21 to 108, and you just put in the RGB color you want. It was a while ago
that I did it, but let me know if you have any issues and I'll do what I
can to help.

Cheers,

Derek

mrdjnash, 2021-08-13:
Just realised that the file I attached to my email didn't come through.
Hopefully the XML below will illustrate what I mean:

<sequence_1>
        <settings>
            <control_number>0</control_number>
            <next_step>1</next_step>
            <sequence_name>Setting Name</sequence_name>
        </settings>
        <step_1>
            <color>Multicolor</color>
            <color_1>0, 0 , 255</color_1>
            <color_range_1>21, 21</color_range_1>
            <color_2>0, 0 , 255</color_2>
            <color_range_2>22, 22</color_range_2>
...

</step_1>
    </sequence_1>

EX8F22, 2021-08-13:
Thank you, Derek! Is this XML something that I can copy & paste into the
software somehow? I have a very limited understanding of this...

closed 2022-11-02 by onlaj