onlaj/Piano-LED-Visualizer #203: Remove the led space

Gregory-Hepicloud, 2021-03-11:
Hi we have small piano 61 keys and 1m led with 100 led i want to put one led by key but on play we have 1 led off for make space but who disable it ? 

thx you

EmanuelFeru, 2021-03-11:
It is basically this function you need to change:
https://github.com/onlaj/Piano-LED-Visualizer/blob/f28b1b19c5dfe49656af7257072d3be0488d845f/visualizer.py#L1638-L1647

In a regular size piano of `88` keys the code above maps the piano notes to LEDs as follows:
```
Notes on piano: [21... 108] --> LEDs on the strip: [2... 176]
```

If you want to light-up Led next to Led, I think you need to change the function above to:
```python
def get_note_position(note):
    note_offset -= ledstrip.shift
    return  (note - 20) - note_offset
```
You might need to adjust the `20` value, if the first note of you piano is **not** `21`.

Gregory-Hepicloud, 2021-03-11:
Hi thx you its solved is need to be this :
`def get_note_position(note):
    if(note > 72):
        note_offset = 33
    elif(note > 35):
        note_offset = 32
    else:
        note_offset = 31
    note_offset -= ledstrip.shift
    return  1*(note - 20) - note_offset` // change 2* to 1* and note offset

Gregory-Hepicloud, 2021-03-11:
And did you know for invert led ? led start form left can not start from right ?

EmanuelFeru, 2021-03-11:
Do you need it the other way around? You can map it from right to left then.

Gregory-Hepicloud, 2021-03-11:
> Do you need it the other way around? You can map it from right to left then.

I am right handed and the cable is too short to start the led strip on the other side did you know I can reverse the direction of the mapping?

EmanuelFeru, 2021-03-12:
To invert led mapping, for the original code is:
```python
return (ledstrip.led_number + 2) - 2*(note - 20) - note_offset
```
while for you would be:
```python
return (ledstrip.led_number + 1) - (note - 20) - note_offset
```
assuming you already set the `ledstrip.led_number = 61` for your case.

Gregory-Hepicloud, 2021-03-14:
Thx you :) its fixed we close issue :)

closed 2021-03-14 by Gregory-Hepicloud