onlaj/Piano-LED-Visualizer #206: no show led display and no works

omegadw, 2021-03-14:
I follow all instructions and show me the error when I login

#Traceback (most recent call last):
  File "/home/Piano-LED-Visualizer/visualizer.py", line 5, in <module>
    import LCD_1in44
  File "/home/Piano-LED-Visualizer/LCD_1in44.py", line 28, in <module>
    import LCD_Config
  File "/home/Piano-LED-Visualizer/LCD_Config.py", line 38, in <module>
    SPI = spidev.SpiDev(0, 0)
FileNotFoundError: [Errno 2] No such file or directory

EmanuelFeru, 2021-03-14:
Did you enabled the SPI interface?
https://www.raspberrypi-spy.co.uk/2014/08/enabling-the-spi-interface-on-the-raspberry-pi/

omegadw, 2021-03-14:
no SPI Interface is not on, thanks , a basic problem, thanks

closed 2021-03-14 by omegadw