onlaj/Piano-LED-Visualizer #208: No LCD hat

MadhavMantri28, 2021-03-18:
Hello there!
I have coded it. However I do not have the LCD hat and as far as I know I wont be able to get it in my area anytime soon. I have 3.2 inch waveshare LCD hat but I dont think that is compatible. I wanted to know how can I handle the LED without the LCD hat. 
Thank you.

onlaj, 2021-03-19:
Hello.
For color mode instead of "Single" you put "Rainbow" or "Speed" or "Multicolor".
Currently it's not possible to start recording midi without lcd hat.

MadhavMantri28, 2021-03-20:
One more thing do I need to reboot the pi every time I make changes in the settings file?

onlaj, 2021-03-23:
No, but you have to restart the script.

MadhavMantri28, 2021-04-01:
Hello. I just got my LCD hat and its working fine. However, I am just able to use the bottom and the up key. The jpress and right left and other key 1 2 3 are not working. Can you please help me here?

onlaj, 2021-04-01:
Try to reconnect the hat, make sure all pins are inserted securely.

MadhavMantri28, 2021-04-01:
Yes I did everything and I have pinned them securely but I dont think this is of any help. I mean I changed the pin number in the code for down. For example if down arrow key was 16 and jpress was 13, I flipped them and found that jpress is working too. So this tells my LCD hat is fine. I dont know.

onlaj, 2021-04-01:
Get demo code and check if it works there https://www.waveshare.com/wiki/File:1.44inch-LCD-HAT-All-Code.7z

MadhavMantri28, 2021-04-01:
I am really sorry but I dont get what I have to do. Can you please elaborate. Thank you very much.

onlaj, 2021-04-01:
Download demo code, unzip, go to RaspberryPi folder and copy paste Python folder somewhere to your RPi, for example to /home/ catalog. Then run the key_demo.py

MadhavMantri28, 2021-04-01:
[image]

It is showing me this error.

onlaj, 2021-04-01:
Run it with python instead of python3 or add parentheses to every print.

MadhavMantri28, 2021-04-02:
I did run with python it is showing me import error : no module named LCD_1in44

onlaj, 2021-04-02:
Is file LCD_1in44.py in the same folder as file key_demo.py?

MadhavMantri28, 2021-04-02:
[image]
I am getting this when I run the program and use my lcd hat.

onlaj, 2021-04-02:
So it works with demo code, that's really weird.
Starting from this [line](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/visualizer.py#L2378) you can find code responsible for detecting key presses. As you can see it looks the same for every key (excluding key 2 and key 3 which has additional functions) and I have no clue why only two of them are working.

MadhavMantri28, 2021-04-02:
I did the whole coding again too yesterday and the issue remains the same.  Can I do something like this 
Since my down key and up keys are working, can I change the code of down key and put some other function?

closed 2021-08-16 by onlaj