onlaj/Piano-LED-Visualizer #214: Webinterface

vzoltan, 2021-03-25:
Hi @onlaj  and @EmanuelFeru  (i see you are very active with the development, thx for you too)

Are you guys planning to make a webinterface for this project? that lcdscreen is sooo tiny;) 
also since we use our laptop / tablet for Synthesia, it would be very convinient to controll the pi (the visualizer menu) via Web.
what do you think? 

Thanks

EmanuelFeru, 2021-03-25:
Hi @vzoltan,
I think your enhancement proposal is interesting. I am not sure how many people have a laptop turned on most of the time with the Visualizer. However, for sure everyone has a smartphone nowadays which can be used for this purpose. This could be even an alternative to make the LCD screen optional. So, a web interface is for sure nice and useful.
Maybe @onlaj has some ideas about possibilities, or what can be done in this respect. Because I started with Raspberry pi programing just recently, so I am not sure what are the capabilities of these Raspberry pi devices.

vzoltan, 2021-03-25:
I already run lighthttpd as webserver to easily sesrch midis and play them on my piano from web using laptop/tablet/phone

Im sure it can handle nginx webserver too.

So this is not a question that the pi is capable for this;)

I just cant program python to work with html/css.

onlaj, 2021-03-25:
It's on my to-do list, but I'm not sure if there is a demand for something like this.

vzoltan, 2021-03-25:
Bc they didn't think about it yet;)
Once its done, im sure everyone will use it :)
But i get it, thanks anyway

AntwanR942, 2021-04-03:
I was thinking about trying to implement this myself but I haven't worked with python much and it would probably take me a while to get my bearings however I definitely think there is demand for something like this!

onlaj, 2021-07-21:
Web interface is merged to master branch :)

closed 2021-07-21 by onlaj