onlaj/Piano-LED-Visualizer #218: Need some help.

AntwanR942, 2021-04-02:
To cut a long story short I had this working but due to the way I positioned it the wires snapped off the LED strip so i purchased a new strip connected it up and nothing is happening. I completely reinstalled the OS and the visualizer back onto the RPI but still nothing. I tried selecting an animation but nothing happens when i click the button. Help would be greatly appreciated. (I have taken some photos, please check them about to see how everything is wired and if you can see anything wrong.)

[image]
[image]
[image]

AntwanR942, 2021-04-03:
I would be willing to replace the RPI but only if someone can assure me that it is the controller that is broken. I am thinking this because randomly the (~7 leds) lights all went white but still not responding to commands. From what I've seen this is a "timing" issue where the WS2812B strip requires very specific timing and thus isn't responding as expected but it also may be a faulty connection somewhere.

AntwanR942, 2021-04-03:
Also, I could measure 5V on the LED strip.

AntwanR942, 2021-04-04:
I've ran some tests in the tests directory and measured some values while they were running. I could still measure 5V across the LED strip as usual. I could measure a fluctuation of voltage from 0V to ~2V from the 2 GPIO pins that are connect to the strip however there was no voltage across the middle and negative contacts actually on the strip. Does this suggest that a faulty connection is not allowing voltage to get to the strip or is there meant to be nothing across the middle(data) negative contacts?

bobsmith1950, 2021-04-07:
I've just started my LED project and had lots and lots of issues to start with.

Now it's behaving itself - I added a resistor on the data line (~470 ohm) and a capacitor across the power connector + / - terminals (1000uF - 6.3V to 25V) - this has cleaned up a lot of my problems.

Also the connections to GND and PIN18 (data signal) on the PI Zero can easily break/short - so i added a GPIO expander to give me space to use jumpers instead.

closed 2021-04-07 by AntwanR942