onlaj/Piano-LED-Visualizer #221: Traceback issues after enter the following command 'sudo python3 visualizer.py'

clancel, 2021-04-09:
Hi.  When I enter the 'sudo python3 visualizer.pycommand', it returns the following script,

Traceback (most recent call last):
  File "visualizer.py", line 5, in <module>
    import LCD_1in44
  File "/home/Piano-LED-Visualizer/LCD_1in44.py", line 28, in <module>
    import LCD_Config
  File "/home/Piano-LED-Visualizer/LCD_Config.py", line 38, in <module>
    SPI = spidev.SpiDev(0, 0)
FileNotFoundError: [Errno 2] No such file or directory

Can you help me?  Thank you.

Regards,
Kim

onlaj, 2021-04-09:
Hey.
Did you enable SPI?
https://www.raspberrypi-spy.co.uk/2014/08/enabling-the-spi-interface-on-the-raspberry-pi/

clancel, 2021-04-09:
Hi.  Thank you for a quick response.  Yes.  SPI Interface was enabled previously but I enabled it again.  Afterwards, I tried the command again and it returned a different response than before.

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/usr/local/lib/python3.7/dist-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/usr/local/lib/python3.7/dist-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "visualizer.py", line 5, in <module>
    import LCD_1in44
  File "/home/Piano-LED-Visualizer/LCD_1in44.py", line 31, in <module>
    import numpy as np
  File "/usr/local/lib/python3.7/dist-packages/numpy/__init__.py", line 145, in <module>
    from . import core
  File "/usr/local/lib/python3.7/dist-packages/numpy/core/__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.7 from "/usr/bin/python3"
  * The NumPy version is: "1.20.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libf77blas.so.3: cannot open shared object file: No such file or directory

clancel, 2021-04-09:
I checked the troubleshooting tips on the numpy link provided and followed the instructions on the troubleshooting page for raspberry pi and now it's returning this,

Traceback (most recent call last):
  File "visualizer.py", line 8, in <module>
    from PIL import Image
  File "/usr/local/lib/python3.7/dist-packages/PIL/Image.py", line 109, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

clancel, 2021-04-09:
I went through the commands from the beginning and this is the latest return after typing in the 'sudo python3 visualizer.py' command. 

pi@raspberrypi:~ $ sudo python3 visualizer.py
python3: can't open file 'visualizer.py': [Errno 2] No such file or directory

Can you help me?  Thank you.

onlaj, 2021-04-09:
You need to be in the right folder to run script.
Type `cd /home/Piano-Led-Visualizer` first.

Didn't you skip something? Like 
`sudo apt-get update`
`sudo apt-get upgrade`
at the beginning, both are important.

clancel, 2021-04-09:
No. I didn't skip anything.  I typed the command, 'cd /home/Piano-Led-Visualizer' and it returned this,

pi@raspberrypi:~ $ cd /home/Piano-Led-Visualizer
bash: cd: /home/Piano-Led-Visualizer: No such file or directory

onlaj, 2021-04-09:
Well, you don't have /home/Piano-Led-Visualizer directory, so either you skipped
`cd /home/`
`sudo git clone https://github.com/onlaj/Piano-LED-Visualizer`
or removed that catalog later.

clancel, 2021-04-09:
Good news!  The visualizer menu is showing on the LCD Hat screen after rebooting.