onlaj/Piano-LED-Visualizer #223: Inconsistent MIDI channel output from Synthesia

answerdon, 2021-04-28:
Dear developers,

I have setup the Synthesia(MAC/IOS) with the RPi via Bluetooth. However, when I run in pratcie mode, I am getting mixed color in the LEDs. In setting, I have selected "finger-based channel" and Omni channel in the output.

Then I tried to look at the MIDI messages going into RPi. It looks like the the MIDI messages are encoded into different channels, even though they are related to the notes on the same hand. For example, Synethesia would use output channel 11 on note 55, 71 & 74, and then change to channel 0, 3... therefore, causing mixed color LED output. Please see below.

[image]

I saw some people have got Synthesisa working with the visualizer on Windows tablets. Is this just an issue with the MAC/IOS version Synthesia? I have tried that on my IPAD and MACbook, both seems to have the same issue. Also, I had no luck with Android.

Have anyone one got Synthesia working with the visualizer on IOS or MAC? Could you please shed some light?

onlaj, 2021-04-29:
What MIDI file do you use? Does the same thing happen with the example files, let's say la campanella from "Songs" folder?
Some midi files has separate channels for each finger

> Channel 0 means "we don't know who or what this note belongs to, but light something up anyway".
> Channel 1-5 is left hand fingers, thumb through pinky.
> Channel 6-10 are right hand fingers, thumb through pinky.
> Channel 11 means "left hand, unknown finger".
> Channel 12 means "right hand, unknown finger".

answerdon, 2021-04-29:
Hi Onlaj,

Thanks for your reply. I have tried to play la campanella in "Learn MIDI" mode. The LEDs turn on correctly in blue and green.

I used the built-in songs in the Synthesia software when I encountered the above issue. In the App GUI, I can see the left and right hand notes are assigned to LEFT(BLUE) and RIGHT(GREEN) respectively.

However, in the MIDI message being parsed, not only multiple MIDI channels are used other than ch11&12, but also the same channel is assigned to the left and right hand note on the same beat sometimes.

closed 2022-11-02 by onlaj