onlaj/Piano-LED-Visualizer #224: Led strips wont turn on

JustTomMcGee, 2021-04-29:
I extended the wires with a pair of 3-pin connectors but now the LED strip wont turn on, unlike before.
RPi still boots up and reads the system image normally.
Is there any chance that the LED strip could be damaged? Also are there other possible issues?

JustTomMcGee, 2021-04-29:
I bought the 3-pin connectors off Aliexpress so I don’t know if the wires could have any problems (mainly because their cheap)

bobsmith1950, 2021-05-02:
checked resistance of each of the 3 wires from the led strip to their ends?
If the only thing you've done is extend them then start with that - could be bad soldering or a broken wire.

JustTomMcGee, 2021-05-26:
I bought a new LED strip AND a new Raspberry Pi but the LED strip still won't turn on
I cut the 3 pin connector from the other end of the LED strip to connect to RPi (I cut at the right end where the arrow points to)
it's strange to me because in the video the 3 pin connector works fine
I'm suspecting the wires but do wires nicked from stripping stop working properly? (white wires connected to 3rd pin and green to 6th, as instructed) 
I don't even know what's wrong with it anymore

onlaj, 2021-05-26:
Can you show us some photos of wiring?

JustTomMcGee, 2021-05-26:
[image]
[image]

Rhynden, 2021-11-30:
Do you still need help? 
Seems you haven't connected the 5V input to the bottom leftmost pin in your first picture.
So you need another cable connecting the + from the adapter to the bottom leftmost pin.

closed 2022-11-02 by onlaj