onlaj/Piano-LED-Visualizer #225: Can't navigate through the menu and LEDs don't work

Stefan97a, 2021-05-01:
[image]

I can't navigate through the menu. I tried to write the file again on the SD card, still the same problem. Does anyone know how to fix this?

onlaj, 2021-05-01:
Hello.
Does the led strip works?
If not, connect to your RPi with ssh (or connect HDMI to monitor if you can), kill the script with

>sudo pkill python3

navigate to visualizer.py folder,

>cd /home/script

run visualizer

>sudo python3 visualizer.py

and show me the output.

If there are no errors, download [demo ](https://www.waveshare.com/wiki/File:1.44inch-LCD-HAT-All-Code.7z) code for Waveshare screen, unzip, go to RaspberryPi folder and copy paste Python folder somewhere to your RPi, for example to /home/ catalog. 

Then run the key_demo.py with
>sudo python key_demo.py

When demo code is running you can try to press keys, their corresponding names should appear in the console.

Stefan97a, 2021-05-01:
Hello, the LED strip worked at the beginning. After I clicked on reboot in the menu it didn't work anymore and the screen kept being white. The next day I turned the Rpi on and it was stuck in the menu where I can't navigate.

bobsmith1950, 2021-05-02:
you need to do the demo test that onlaj posted above - it could just be the waveshare hat playing up rather than a code issue.

Stefan97a, 2021-05-03:
I can't connect my Rpi with ssh, it just doesn't show up with the IP address. Would a new SD card maybe be a solution?

Stefan97a, 2021-05-06:
[image]

Now I tried it with a new SD card, still the same problem. When it turns on, there is like 1 second before it clicks by itself and then I can't navigate.
If I try the LED strip without the LCD hat, it doesn't work. Could it be the Rpi?

bobsmith1950, 2021-05-06:
can you plug the pi into a monitor and maybe hook up a usb keyboard to it? then you can test the rpi directly.

Stefan97a, 2021-05-13:
[image]

I connect the Rpi to the monitor and put in your commands

onlaj, 2021-05-13:
My bad, I had wrong command copied.
Instead of
`cd /home/script`

use
`cd /home/Piano-LED-Visualizer`

But from your screenshot I can see that script doesn't return any errors, so try to run demo file.

Stefan97a, 2021-05-13:
[image]

Now I get this and nothing happens when I press the keys on the piano

onlaj, 2021-05-13:
It shouldn't, you are suppose to check if the screen's keys are working.
Did you download and upload demo files to your RPi?

Stefan97a, 2021-05-13:
I uploaded the demo files to the Rpi.
No reaction from the screen keys

onlaj, 2021-05-13:
So I assume that either the pins on the raspberry pi are damaged or the screen itself.

Stefan97a, 2021-05-13:
I think it might be the Rpi then, because before it gets stuck in the menu, I can navigate for a short moment. The LED should at least work without the LCD hat right?

Stefan97a, 2021-05-20:
I bought a new Rpi and now I can navigate through the menu again and also control the LED animations. But now my problem is that the LEDs don't react to the keys on the piano

closed 2021-10-08 by onlaj