onlaj/Piano-LED-Visualizer #227: randomly freezes

literallyjustbeeb, 2021-05-19:
as im playing a song the raspberry pi randomly stops working, the screens stays the same, but no buttons work, the led lights also stop responding to the keys, though they do stay the same (like if they were on an animation it would stop but also stay on)

onlaj, 2021-05-20:
Hello.
Does the console say anything?
Connect to RPi through SSH, stop all python processes
` sudo pkill python3`
and run visualizer
`sudo python3 /home/Piano-LED-Visualizer/visualizer.py`
play on piano until it freezes again.

literallyjustbeeb, 2021-06-27:
i actually noticed the reason, its totally ok now, it was apperently i set the led count too low, and it started freezing when i pressed the notes that those led corresponded to. thanks though, and no it just stayed the same screen and didnt move at all

closed 2021-06-27 by literallyjustbeeb