onlaj/Piano-LED-Visualizer #228: Is this return normal after booting up the pi with the Visualizer.py code in /etc/profiles?

LoganOrr-dev, 2021-06-03:
Is this return after booting up the pi normal? What concerns me is the import error, is this because I have not connected the Piano or LEDS?
[image]

Logan-Orr, 2021-06-03:
And as I press enter, it says this under it,

[1]+  Exit 1                  sudo python3 /home/Piano-LED-Visualizer/visualizer.py

onlaj, 2021-06-04:
You probably installed wrong version of mido. 
Did you use `sudo pip3 install -r requirements.txt` command?
Did it work earlier? I need more information on that.

Logan-Orr, 2021-06-04:
Well, if i use the command you gave me, it just returns that i already have the package

This did not work earlier, even if I run the code manually, it says the same thing

Logan-Orr, 2021-06-04:
By the way, I am using a raspberry pi 4

Logan-Orr, 2021-06-04:
It says that all the packages are updated and satisfied

Logan-Orr, 2021-06-04:
Was I supposed to use that command? I did not see it in the manual installation page

onlaj, 2021-06-04:
I assume you are using system image from [releases](https://github.com/onlaj/Piano-LED-Visualizer/releases)?

Logan-Orr, 2021-06-04:
No, i used manual installation

Logan-Orr, 2021-06-04:
Now i get this 
[image]

Logan-Orr, 2021-06-04:
What do I do now?

onlaj, 2021-06-04:
Now it seems there is a problem with rpi_ws281x package, maybe it's damaged, try 
`sudo pip3 uninstall rpi-ws281x`
`sudo pip3 install rpi-ws281x`

you can try to reinstall all packages, just to be sure
`sudo pip uninstall -r requirements.txt -y`
`sudo pip3 install -r requirements.txt`

Logan-Orr, 2021-06-04:
Ok, heres what i got
After uninstalling and installing
[image]

Logan-Orr, 2021-06-04:
[image]

onlaj, 2021-06-04:
Did you install only rpi-ws281x package or all of them?
If reinstalling all of them doesn't help I would try to start everything from scratch. Maybe use release image?

Logan-Orr, 2021-06-04:
All of them

Logan-Orr, 2021-06-04:
But i want to acsess ssh through it

Logan-Orr, 2021-06-04:
I dont belive you can do that on the realse image

Logan-Orr, 2021-06-04:
If you can, please send me the link to the dowload

onlaj, 2021-06-04:
You can https://github.com/onlaj/Piano-LED-Visualizer/releases/tag/v1.00

Logan-Orr, 2021-06-04:
If I try to use the one in realeses, the zip file is so big that you cant unzip it

Logan-Orr, 2021-06-04:
So what should I do

Logan-Orr, 2021-06-04:
Cause i use macos

onlaj, 2021-06-04:
It's only 2.4GB after unzipping.

> Can you use the one from source forge?

I didn't make that one, so I can't vouch for it.

Logan-Orr, 2021-06-04:
But if gives me error 79 inappropriate file type or format on yours

Logan-Orr, 2021-06-04:
Can you give me the standalone img file unzipped?

Logan-Orr, 2021-06-04:
Or like a google drive download link

onlaj, 2021-06-04:
Unzipping works on windows and linux, I didn't test it on macos, I don't have any apple products.
I can unzip and host uncompressed version somewhere, but not today, I need several hours.

Logan-Orr, 2021-06-04:
So, what should i do

Logan-Orr, 2021-06-04:
Hmmm

Logan-Orr, 2021-06-04:
Is there a software i could use?

onlaj, 2021-06-04:
I don't know, as I said, I don't have any apple products.

Logan-Orr, 2021-06-04:
Ok, got it, Download WinZip for mac then unzip the file, after that drag the imp file to your downloads and you have the imp file, I would edit the repo page so others don't have this problem

Logan-Orr, 2021-06-04:
how do I ssh into it

onlaj, 2021-06-05:
The same way you ssh into your own image.

closed 2022-11-02 by onlaj