onlaj/Piano-LED-Visualizer #230: LED Shifting

LeonRappelt, 2021-06-18:
I have the Problem that the leds start to shift going out from the middle. I tried to shift the the Led and adjust the count but still are some Leds out of syn with the keys

onlaj, 2021-06-18:
Hey.
How it looks when you launch animations, let's say "Scanner"?
Try to launch tests/printmidimessages.py and play lowest and highest key on your piano, what does the script output?

AntwanR942, 2021-07-23:
Hey, I had this exact problem with my keyboard. I made an edit to the `get_note_position` function in `functions.py` in the `lib` folder. I could make a PR but this may affect already existing pianos where the lights are lined up fine. 
```py
def get_note_position(note, ledstrip):
    note_offset = 0

    if note > 27:
        note_offset = math.ceil((note - 27)/14)
    note_offset -= ledstrip.shift
    note_pos_raw = 2 * (note - 20) - note_offset
    if ledstrip.reverse:
        return max(0, ledstrip.led_number - note_pos_raw)
    else:
        return max(0, note_pos_raw)
```

AntwanR942, 2021-07-23:
You'll also need to import the math module and also if you notice that it isn't offset well you can change the `14` in `(note-27)/14` to change how often the led is offset. A lower number means the led is offset more often - `13` works really well for me.

Joker-rider, 2021-11-14:
Hello
Can you give more details about your modification. Because it blocked my raspberry, I have awhite screen and nothing is working. Thank you so much.

closed 2022-11-02 by onlaj