onlaj/Piano-LED-Visualizer #231: Issues with connection(?)

ar5ts67, 2021-06-20:
I recently made this project and everything seemed to be going well until I started practicing with it. Some notes are fully skipped over, both the lights and sounds. Starting up is ridiculously complicated when it comes to connecting the pi and computer to each other. I usually don't even bother trying to get the light bar to work because its such a hassle and usually just skips half of the notes I need it to display. I'm 100% sure this is user error, I have no clue what I'm doing beyond following the instructions.
Hardware I'm running is the Pi and LCD hat recommended, my piano is a Williams Allegro 3, and the computer I'm using is a Surface Pro 3 running Windows 10. I'm using a Sevilla Soft Midi to USB, the same one detailed in the instructions.
My settings in Synthesia are:
**Input**
Synthesia Virtual Piano - off
Midi USB-USB (1 through 4) - on
**Output**
Built-in MIDI synthesizer - off
MIDI USB-USB (1-4) - Everything, Keylights on
Microsoft GS Wavetable Synth - Everything
[image]
[image]
Thank you for any help, I'm really hoping to use this as training wheels for getting back into piano.

onlaj, 2021-06-20:
Hey.
The first thing that caught my eye is that sevilla usb-usb should be between your PC and USB HUB, not piano and hub.

ar5ts67, 2021-06-20:
I 100% labeled it incorrectly, the Sevilla Soft is between the computer and the hub, the other USB connection in the Hub is to the piano.

onlaj, 2021-06-20:
I assume there is no problem with lighting when you set input to your piano, it registers every key?
Try to run "free play" in synthesia and check if it skips any notes there.

ar5ts67, 2021-06-20:
The only issue I have with lighting on free play is some of the lights stay on instead of fading. I currently have the rpi's input set to USB midi 1 and the playback on the piano. I think I originally put all of the MIDI channels on because I was having an issue with Synthesia not detecting the rpi when I would start a practice session, but that issue seemed to be just because I wasn't restarting Synthesia, _I think_
Free play also registers on the Synthesia just fine, all the key presses, but I'm finding that some keys tend to hold as if I'm on the sustain pedal even when Im not. When that happens the light for that key stays lit until I strike the key again.

Orly1625, 2021-08-25:
Hello, I would like to know if the problem with the midi interface has been solved, so that I can buy one without fear of making a mistake in its operation, thanks for the answer.

closed 2023-09-11 by onlaj