onlaj/Piano-LED-Visualizer #239: Playback Funktion does not work with LEDs

justkatia, 2021-07-19:
Hello, thank you very much für sharing your project I‘m almost done. When I start the Playback function for example the Campanella.midi file the leds doesn‘t light up. Is there any way how i could change/fix it? Thanks

onlaj, 2021-07-19:
Hey.
Have you set up your piano as a playback port?
Is the piano playing a song?

justkatia, 2021-07-19:
Yes the piano is playing the Song but the LEDs doesn‘t light up

onlaj, 2021-07-19:
Have you downloaded the latest updates? Does the leds light on when you play? Have you tried restarting the raspberry pi?
I'm trying all the options here, I just tested it with my piano and it works just fine, so I doubt I messed it up with the last update.

justkatia, 2021-07-22:
yes I downloaded the latest update and the leds light up when I play. the only problem is the playback function. The song is played but the leds does not light up. I also noticed that when I choose a song from the playback function my pi freezes and I can't do anything in the menu until I restart the pi.

Voidshade, 2021-09-08:
> Yes the piano is playing the Song but the LEDs doesn‘t light up

Hi, I have same issue just the opposite,when I play midi LEDs showing but can't hear music

closed 2022-11-02 by onlaj