onlaj/Piano-LED-Visualizer #240: "Learning to play with Synthesia" help on tablet

nicksource, 2021-07-29:
I've got everything setup and working except for learning to play, which I don't quite understand the way to connect my android tablet to it using cables (which I assume is the best option here).

Can someone explain the cables required and how these are connected together?

It suggests a USB Midi device but it has 2x USB ends I don't get what goes where with this. I also cannot find the correct light support settings I need to change in the Synthesia app.

If anyone has got this setup and working some help would be much appreciated, or alternatively how you have it linked to your tablet for this feature to work. Thank you.

onlaj, 2021-07-29:
Hello.

>It suggests a USB Midi device but it has 2x USB ends I don't get what goes where with this.

You put it between your PC/tablet and RPi. I made a simle diagram, it should clear things up.

[image]

nicksource, 2021-07-29:
Thank you so much, so in the instance of my Android tablet, it would be a USB-C (tablet end) to USB-A (into the USB device). Then a USB-A to USB-A from the midi device to hub.

What does the MIDI USB device actually do? Why can't I just connect directly to the USB hub, does it encode/decode something?

onlaj, 2021-07-29:
>What does the MIDI USB device actually do?

It emulates midi device. 
It is possible to make a direct connection with just USB cables, but some additional software on both ends would be needed.

closed 2021-08-16 by onlaj