onlaj/Piano-LED-Visualizer #242: Brightness changing issue

DjangoPiano, 2021-08-07:
Hi,
Thank you very much for your work and espacially for your update with the user interface. All works fine for me except the changing brightness feature on the web interface. Once I try to change it, all the system freezes : the rpi and the leds.

Any ideas ?

Your help will be much appreciated.

Thank you very much for your consideration.

onlaj, 2021-08-08:
Hey.
Can you SSH into your rpi?
Type those commands:

`sudo pkill python3`
(it will stop all python scripts running)

`sudo python3 /home/Piano-LED-Visualizer/visualizer.py`

then try to change brightness and check if it printed any errors in console.

Gauthiermayer, 2021-08-11:
Hello sir, I can confirm that there is no printed error.
When changing quickly the brightness the python app is still running but the led strip is stuckor bugged and the API always gives `{"success":true}`.
I tried different ways to provoke the bug but It seems that it appears randomly

onlaj, 2021-08-11:
@DjangoPiano  @Gauthiermayer 
Although I can not reproduce this error on my side, I will try to make some changes. Currently, when changing led strip brightness, the script recreates the object and reinitializes the library. I checked the available functions of the library and it is possible to change the brightness of the existing object.

I will let you know when I push the commit.

Gauthiermayer, 2021-08-11:
@onlaj @DjangoPiano 

https://github.com/onlaj/Piano-LED-Visualizer/blob/daf4aa0ccb804a6b1e6d9b9b1138d3addc61f4da/neopixel.py#L127

Maybe using this function from neopixel lib would be better

Something like `self.strip.setBrightness(self.brightness)` instead of 
https://github.com/onlaj/Piano-LED-Visualizer/blob/daf4aa0ccb804a6b1e6d9b9b1138d3addc61f4da/lib/ledstrip.py#L47-L48

Gauthiermayer, 2021-08-11:
I solved it ! 

Use `self.strip.setBrightness(int(self.brightness))`
instead of 
https://github.com/onlaj/Piano-LED-Visualizer/blob/daf4aa0ccb804a6b1e6d9b9b1138d3addc61f4da/lib/ledstrip.py#L47-L50
I can do a pull request if you want

onlaj, 2021-08-11:
Did you test it? 
The same can be applied to `change_led_count` I suppose. 
I will test it tomorrow and then merge.

Gauthiermayer, 2021-08-11:
Yes and it works! 
I also searched for `change_led_count` but I didn't find any function tu use instead

closed 2021-08-11 by DjangoPiano