onlaj/Piano-LED-Visualizer #243: Flask

Logan-Orr, 2021-08-09:
So, this project has been working fine, but one day i turned it on and it did not boot, it was stuck at the white screen, I reinstalled the software, it worked. But, I still had to update it, I updated it and tried to boot it, same thing. Did not work. I was able to ssh into it and it said the it was missing something called Flask, I did not do anything to the software in any way that would make it stop working, please help.

Here is the error after updating: 
[image]

onlaj, 2021-08-09:
Hello.
Sorry for this inconvience, but in latest updates I added a lot of new features which requires additional libraries. You can install them using command:

`sudo pip3 install -r requirements.txt`

Logan-Orr, 2021-08-09:
Do I reinstalled then do this, cause it gives me this error
[image]

onlaj, 2021-08-09:
You need to be in visualizer directory

`cd /home/Piano-LED-Visualizer/`

Logan-Orr, 2021-08-09:
Ok

Logan-Orr, 2021-08-09:
Now this;
[image]

Logan-Orr, 2021-08-09:
[image]

Logan-Orr, 2021-08-09:
Maybe I installed the wrong version?

Logan-Orr, 2021-08-09:
Should I try to restart?

onlaj, 2021-08-09:
What did it show after `sudo pip3 install -r requirements.txt`?

Logan-Orr, 2021-08-09:
It installed everything, no error

Logan-Orr, 2021-08-09:
Including flask

onlaj, 2021-08-09:
Try to restart then, you can check installed packages with
`pip3 list`

Logan-Orr, 2021-08-09:
Like reintall and stuff

onlaj, 2021-08-09:
For now just restart RPi, I doubt it will help but it's worth a try

Logan-Orr, 2021-08-09:
K

Logan-Orr, 2021-08-09:
Yeah did not work

Logan-Orr, 2021-08-09:
Would this mean that it will not work for everyone else?

onlaj, 2021-08-09:
Works for me, other users also tested latest update and no one reported problems with importing flask module.

I will soon make new release with everything configured.

Logan-Orr, 2021-08-09:
Should I re download the image of the github page and use that have you updated the image recently

onlaj, 2021-08-09:
No, image is not updated.
btw. try to 
`sudo pkill python3`
before launching new script. When you ssh it autostart script, maybe you have moer scripts running and it can cause some problems.

Logan-Orr, 2021-08-09:
Where am I supposed ptut this

Logan-Orr, 2021-08-09:
You can just tell me the exact instructions or walkthrough of this like what commands to enter and when

onlaj, 2021-08-09:
SSH to your RPi, then go into visualizer directory
`cd /home/Piano-LED-Visualizer/`

stop all python scripts, just in case
`sudo pkill python3`

start script
`sudo python3 /home/Piano-LED-Visualizer/visualizer.py`

Logan-Orr, 2021-08-09:
Did not work

onlaj, 2021-08-09:
What does it show? Still the same error?

Logan-Orr, 2021-08-09:
same error

Logan-Orr, 2021-08-09:
I just installed a fresh software and its booted, and still on the old version, so no learn midi etc

Logan-Orr, 2021-08-09:
what do I do now

onlaj, 2021-08-09:
Well, I can only recommend that you reinstall everything (maybe try to follow manual setup?) or wait till I release new image.

I'd like to finish the webinterface first tho; hopefully, I'll be able to do it this week.

Logan-Orr, 2021-08-09:
I GOT IT, what you have to do is run `sudo pip3 install Flask` before you update, and then right after you update, you have to install the requirements again before you reboot

Logan-Orr, 2021-08-09:
oh and the web interface is cool! Keep working on it

closed 2021-08-09 by Logan-Orr