onlaj/Piano-LED-Visualizer #251: booting from SD card reliably

macaroniart123, 2021-08-21:
I have finally gotten everything to work perfectly... except. 

After I load the program on my raspi, I can turn it off and on once or twice, but then if I try to boot the program from the sd card the next day I don’t get anything at all on the display. I shut it off properly from the visualizer menu. 

I burned an image of the program when it was working, and if I copy that img onto a SD card - I can get the project to work again. But only once, maybe twice . If I shut it off properly, go away for a few hours, then when I go to turn it back on I get nothing.  It seems I have to reburn the sd every time I would want to play the piano.? Does anyone know what may be causing this or I can do to fix this?

Help is very, very appreciated!!!!

onlaj, 2021-08-21:
Does it work when you try to connect to it through ssh?

Logan-Orr, 2021-08-21:
Does it show a white or black screen? There’s a bug where if you leave it off for a few hours, when you try to turn it on the screen goes black after a few seconds of showing the menu, I just touch the keys on the piano and it wakes back up, or turn it off and on

macaroniart123, 2021-08-23:
> Does it work when you try to connect to it through ssh?

I am not able to connect via ssh when this occurs. The Raspi IP address does not show up as connected to my wifi on my ip scanner

macaroniart123, 2021-08-23:
> Does it show a white or black screen? There’s a bug where if you leave it off for a few hours, when you try to turn it on the screen goes black after a few seconds of showing the menu, I just touch the keys on the piano and it wakes back up, or turn it off and on

It is illuminated- still dark but you can tell it is emitting light.

thanks for the response!

onlaj, 2021-08-23:
Can you install Raspberry OS lite on SD card, without any Visualizer software, just clean system? After that try to do the same, power it off and leave it for some hours, then try to boot again.

DjangoPiano, 2021-10-31:
Hi Macaroniart123,
I have exactly the same issue as yours... Do you have any update on it ?

Thank you.

closed 2023-09-11 by onlaj