onlaj/Piano-LED-Visualizer #253: I'm a very newbie. Help me plz..

songazi123, 2021-09-02:
There was an extra raspberry pie 3B left in my house, and I proceeded with it, But it didn't work at all. (If Raspberry Pie Zero WH is absolutely necessary, I will repurchase it.)
I tried manual installation too, but I failed because I'm a beginner.
I think this problem has occurred because I connected the wrong wire or did not solder.
And I don't have a 1.4 inch LCD.(If this is necessary to proceed with the project, I will get it. / I just need the LED function to react when I press the piano.)
I'd really appreciate it if you could help me with my problem. (I am sorry that I am not good at English :) 
[image]
[image]
[image]
)

onlaj, 2021-09-02:
Hey.
Your wiring looks wrong, check this site: https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/
There are 5 cables coming from led strip but in this setup we only use red green and white set, you can ignore the red and white two-cables set.

songazi123, 2021-09-03:
black and green wire is 7inch LCD display's one. I connected the two lines of the LED strip to GPIO18 and GND, and the + pole to the DC socket.
[image]
I also noticed that LED strips were occasionally briefly lit by about a fifth. (or just one last LED).

onlaj, 2021-09-06:
GND should be connected both to RPI and led strip, you need to put led strip's GND wire to socket.

closed 2022-11-22 by onlaj