onlaj/Piano-LED-Visualizer #260: Wrong LCD

Moofish87, 2021-09-19:
Hi, 
I unfortunately ordered the wavshare 1.3 240x240 lcd. I can see someone has written code to support this screen but is this supported on the image file? I’ve loaded the image file and the hat is not displaying anything. Or do have to manually install the software? If it’s the manual option is there a guide anywhere for Pi that’ll help me 

thanks

onlaj, 2021-09-20:
Hey.
Connect to your RPi through ssh and follow the last step, starting from `Enable autostart script on boot:`
https://github.com/onlaj/Piano-LED-Visualizer/blob/master/Docs/manual_installation.md#7-installing-piano-led-visualizer

Moofish87, 2021-09-28:
Thank-you soo much for your help with this. I can’t believe I managed to make it work. I have got another question but I’m not sure if it’s better to start a new thread. My led strip for some reason is only lighting up the first 16 LED’s  and a random one in the middle of the strip. It’s there a way of seeing if the strip is faulty or if it’s software.

onlaj, 2021-09-29:
Try to run some led animations. If still only the first 16 LEDs are lighting, I would check the wiring.

Moofish87, 2021-09-29:
I’ve tried running the animations and still only the first 16 lights light up. These lights are still lit even when I disconnect the raspberry pi. Is there a possibility that the police and the led strip are not communicating.

Moofish87, 2021-09-29:
“Pi and the led strip”

onlaj, 2021-09-29:
It's either wiring or a damaged led strip, but you can try to reinstall software just to be sure.
Take a look at the led strip, maybe you can notice some damage.

gstrug, 2022-01-31:
Hi,
I can only find the 128x64 waveshare 1.3 128x64 oled hat. is there a way to get this to work. I have followed the manual install steps.
thanks

onlaj, 2022-01-31:
@gstrug 
This is an OLED, two colors only. It would require more work to make it compatible. If you can't find the right screen you can just instead use web interface to control your Visualizer.

closed 2023-09-11 by onlaj