onlaj/Piano-LED-Visualizer #267: AttributeError: 'rtmidi._rtmidi.MidiIn' object has no attribute 'delete'

dimespi, 2021-11-03:
@onlaj  I've updated the program but now I have this traceback when I start my raspberry: 

```
pi@raspberrypi(ro):~$ Namespace(clear=False, display=None, port=None, webinterface=None)
no input port
no play port
Traceback (most recent call last):
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 85, in <module>
    menu = MenuLCD("menu.xml", args, usersettings, ledsettings, ledstrip, learning, saving, midiports)
  File "/home/pi/Piano-LED-Visualizer/lib/menulcd.py", line 45, in __init__
    self.update_ports()
  File "/home/pi/Piano-LED-Visualizer/lib/menulcd.py", line 122, in update_ports
    ports = mido.get_input_names()
  File "/home/pi/Downloads/mido/mido/backends/backend.py", line 169, in get_input_names
    devices = self._get_devices(**self._add_api(kwargs))
  File "/home/pi/Downloads/mido/mido/backends/backend.py", line 163, in _get_devices
    return self.module.get_devices(**self._add_api(kwargs))
  File "/home/pi/Downloads/mido/mido/backends/rtmidi.py", line 65, in get_devices
    mi.delete()
AttributeError: 'rtmidi._rtmidi.MidiIn' object has no attribute 'delete'
```

It worked fine before the update.

onlaj, 2021-11-03:
Hello
Try to also update all required packages

`sudo pip3 install -r requirements.txt --upgrade`

dimespi, 2021-11-03:
Thanks, works fine now! 

I have another issue, when I touch my midi keyboard the lights don't turn on. Here is my port settings:

[image]

When I try one of these animation it works though:

[image]

I'm using Ableton and the sound is working when I press a key.

dimespi, 2021-11-05:
@onlaj Works fine now, thanks!

closed 2021-11-05 by dimespi