onlaj/Piano-LED-Visualizer #270: La lumière des led non aligner sur certaine touches

Joker-rider, 2021-11-06:
Existe-t-il un moyen d'ajuster la séparation entre les leds pour les aligner avec les touches ? J'ai un petit problème où au fur et à mesure que les touches descendent ou s'élèvent, l'alignement de la lumière dérive. Le C central et quelques autre touches sont parfaitement alignées.
Mon clavier à 76 touches. Config origine.
J'ai essayé d'ajuster avec l'interface web mais à part déplacer les leds et mon C centrale rien ne change.
Impossible de dire que lorsque j'arrive à une certaine touche la led a allumé change.
Exemple : C = do = LED 72+8 =OK
En descendant :
Re# = LED 63 = OK, Re# = LED 39 = KO => LED 40 = OK.
Et montant
Si = LED 72+30 = KO => LED 72+29 = OK
[image]
[image]

Merci de votre aide

onlaj, 2021-11-08:
Hello.
It seems unusual, I wonder if the reason is the non-standard width of the piano keys or maybe a different model of the led strip.
I will try to make some option to adjust spacing, but for now you can try to fix it by yourself. The function that is responsible for that is located in [/lib/functions.py](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/lib/functions.py) under 

```
def get_note_position(note, ledstrip):
    if note > 92:
        note_offset = 2
    elif note > 55:
        note_offset = 1
    else:
        note_offset = 0
```

It means that for notes higher than 92 the offset it equal to 2, and for notes higher than 55 (but lower than 92) it's 1. You can try to adjust those values, change 92 to something lower like 72, and 55 to 35, you need to test it by yourself.

Joker-rider, 2021-11-08:
Hello
Thanks
I will test the solution tonight and give tout a feedback.

Joker-rider, 2021-11-09:
Hello.
Is there a possibility of being able to carry out several offset?
Example : one in note 92, one in note 72, one in note 55 and one in note 35?

Joker-rider, 2021-11-10:
Hello
is there a test mode to light all the leds corresponding to the keyboard notes in order to best place the led strip. the ideal would be to be able to make the adjustments directly with the leds on (number of keys = leds, offset settings in order to correct the drifts ...)

I am still in settings mode to align the leds on my keys.

Bye

Joker-rider, 2021-11-15:
Hello Onlaj,
I would like to know if it is possible to letter more than two offsets (example 4 offsets)?

onlaj, 2021-11-15:
Yes, just edit the code accordingly.

for example:

```
def get_note_position(note, ledstrip):
    if note > 92:
        note_offset = 3
    elif note > 55:
        note_offset = 2
    elif note > 35:
        note_offset = 1
    else:
        note_offset = 0
```

Joker-rider, 2021-11-17:
Hello Onlaj
Have you other solution ?

onlaj, 2021-11-18:
Can you show me how you edited the "get_note_position" function? Screenshot of your text editor will be fine.

Joker-rider, 2021-11-19:
Hello,
here is the screenshot
[image]

Joker-rider, 2021-11-19:
Hello
here is the screenshot

Télécharger Outlook pour Android<https://aka.ms/AAb9ysg>

Joker-rider, 2021-11-22:
Hello Onlaj
do you have solution to my problem ?
Thank you so much

onlaj, 2021-11-22:
Sorry, somehow I missed last response.
Can you launch visualizer (with your changes) manually?

`sudo python3 /home/Piano-LED-Visualizer/visualizer.py`

if it crash post an error message here.

Joker-rider, 2021-11-22:
here is the test
[image]

Joker-rider, 2021-11-22:
Heure is the test
[image]

Joker-rider, 2021-11-22:
Here is the test.
[image]

onlaj, 2021-11-22:
Did you copy the code from [here](https://github.com/onlaj/Piano-LED-Visualizer/issues/270#issuecomment-968690075) or retype it by yourself? 
When editing code, use 4 spaces instead of tabs.

Joker-rider, 2021-11-22:
I used 4 spaces instead of tabs and it worked.
Perfect.
Thank you so much.

closed 2021-11-22 by Joker-rider