onlaj/Piano-LED-Visualizer #272: Request: Random color led

Caioalcantaraa, 2021-11-13:
First, I would like to thank you for your work.

Would it be possible to create an animation to light colors randomly on the leds?

onlaj, 2021-11-13:
Hello.
Can you elaborate? What should it look like?
And do you mean animation on idle or when you play?

Caioalcantaraa, 2021-11-15:
Hi,
I have a video demonstrating how it works:
https://youtube.com/shorts/rR9g6TaUXVc?feature=share

I used a code but it was from Arduino, I'll post the link to which file I used:
Vídeo: https://youtu.be/kAj78IE2704

Arquive “random colora”: https://github.com/roy-sukrit/Arduino-LED-Piano

onlaj, 2021-11-15:
Set color to rainbow, scale and timeshift to 1000 and you have random colors :D

closed 2021-12-17 by onlaj