onlaj/Piano-LED-Visualizer #273: Do I have to restart raspi whenever I turn off my keyboard?

vincentxu123, 2021-11-14:
While the LED visualizer does work for my keyboard, I noticed that if I turn my keyboard off and on, the visualizer does not detect the keyboard inputs and light up unless I restart the raspi. Is this normal? Or is there some way I can have the raspi detect keyboard inputs again even after I turn the keyboard off and on?

onlaj, 2021-11-14:
Hello.
What happens if you select your keyboard as an input device when using a screen or web interface? Does it work then?
Also, try to unplug and re-plug your piano's USB.

alexandrewillame, 2022-09-07:
I have the exact same problem.

if I select your keyboard as an input device when using a screen or web interface e.g. switching port twice (when the piano is ON), it work. But again as soon as I turn the piano off, then on, the system does not work anymore and I need to restart the Pi or manually do that port switching for it to work.

Is there any way it can detect the piano is back up (it does since the input port appear back up) and connect to it properly?

Thanks

Lyliya, 2024-09-19:
Hello,
Same problem here, to avoid the quite long boot time, mi Pi 0 is always on, I just turn off the led power and the piano. Turning back on the piano, the system fail to catch the midi port and display nothing on the led. I have to go to the port section in the web app, and swap the port twice to make the connection again