onlaj/Piano-LED-Visualizer #280: Visualizer does not run after last update

EFeru, 2021-11-19:
Hi Onlaj,

I just updated the Visualizer to the most recent version, did everything, RPi0 update, upgrade, Visualizer reinstall of requirements, reboot. But I still get the error in the picture. Any idea?
[image]

onlaj, 2021-11-19:
Hello.
I will fix it in next update, for now follow this part of manual: https://github.com/onlaj/Piano-LED-Visualizer/blob/master/Docs/manual_installation.md#2-creating-autoconnect-script

closed 2021-11-19 by onlaj

EFeru, 2021-11-19:
It worked! Even though the midi.service file does not exist or I couldn't create it in my case. But it works even without.

Awesome job on the webinterface! I like it very much! Thanks