onlaj/Piano-LED-Visualizer #285: File will not unzip

jjonahjamesonIII, 2021-12-01:
it seems like i must download the software here in order to learn from the vizualizer/synthesia. When unzipping the disc image file regardless of os the same error appears "8000x4005".....is there an intermediate step between downloading the file and unzipping it? if so im completly lost so any hints towards the right direction would be much appreciated. 

The manual connections are all built...tried and true.....but i downloaded different software from a link on youtube. It works for the reactive visualizer bit but thats not what im after at all :.(

is it time to give up and go to a coding shop?

onlaj, 2021-12-01:
What software do you use to unzip the file? Did you try 7zip?

jjonahjamesonIII, 2021-12-08:
Just tried 7zip he that seems to solve the issue. Struggled to get synthesis to work even with the special midi usb. Gonna try loading a song on the drive instead. 

Thanks for the suggestion btw. I was just using default Microsoft software

closed 2021-12-30 by onlaj