onlaj/Piano-LED-Visualizer #287: MIDI Ports and Multiple Visualizers

psscin01, 2021-12-05:
Hi onlaj,

I am a teacher and I am trying to implement this project for my students to both complete and use. Ideally, I would like to make this a project done by our 9-12 grade STEAM students for our Fine Arts students to use with Synthesia. I have attempted to make this prototype myself, but I have had many issues with the ports, connections, and consistency. I am somewhat familiar with MIDI setups and I have some experience with computer music. The prototype I made used a P70s Yamaha keyboard (has a Midi in/out) with a Pi zero and an iPad/Macbook Air. 

My problem is this: although I was able to establish connections via bluetooth and rtpmidi, they were often dropped and/or the keys lights failed to respond or show the next-to-play keys. I think the best solution would to hardwire everything together via USB OTG hub, but I can't seem to find a way to keep the tablet charged this way. Since this would be installed in a piano lab with children, I would probably need to keep them on all the time. 

So far, I can't seem to find a reasonable way to connect the pi and tablet with a usb hub. I know there is the Sevilla Soft USB USB MIDI interface, but I don't know if there would be enough units available for the number I would need (at least 8).  I can also feasibly use Synthesia on monitors with macbook airs and mount those to the keyboard, but I still would have a problem with connections and hubs since most of the available computers only use USB-C ports. Does anyone have any thoughts on this? 

Additionally, I am quiet confused by the visualizer port settings that list the input as
 
Device A : Devise B <-----> Device A : Device B 

And a playback port that doesn't seem to make too much of a difference. I'm just not sure what these ports do.

I followed the automatic MIDI connection on the neuma site and still have a lot of issues with assigning the ports. Any guidance on this would be greatly appreciated. I am also going to post this in the general discussion on the Synthesia boards in case that is part of the problem. Please let me know if you need any other additional information. Thanks!

onlaj, 2021-12-06:
> but I can't seem to find a way to keep the tablet charged this way.

You would need to search for a usb-c hub with Power Delivery port.

> but I don't know if there would be enough units available for the number I would need (at least 8)

You can try to contact them. With that many units, you could even count on some discount. Right now they have 10 units in stock, so it should be enough.

>And a playback port that doesn't seem to make too much of a difference. I'm just not sure what these ports do.

The playback port is only for sound when playing midi files.

When you want to light leds with your keyboard, you set your piano as input. For Synthesia learning mode, you set rtpmidi or usb-usb device. Additionally, on the web interface, you can set a secondary port for a quick switch between them.

psscin01, 2021-12-06:
Thank you so much for getting back to me!  

So that is what the switch button is for! In that case, that is very convenient. Also, I tried a USB-c hub with power delivery, but the iPad said it provided too much power and shut the port off. I may need to try some others to see if there is one that can regulate the power a bit. 

I really appreciate all you have done with this project!

closed 2022-11-02 by onlaj