onlaj/Piano-LED-Visualizer #293: How do you assign a color of your choosing to a scale?

clancel, 2021-12-11:
Hi.  I wish to assign individual colors to a 7-note scale that can be looped throughout the keyboard.  For instance,

C= Red
D=Yellow
E=Green
F=Blue
G=Pink
A=White
B=Orange

I guess I would also have to assign the sharps such as C#, Bb, etc.

I know that I have to choose the rainbow option in order to achieve this.  Can you help me?  Thank you so much.

onlaj, 2021-12-11:
You need to use multicolor mode, just set a color for each key.
There is no support for coloring specific key in scale yet.

clancel, 2021-12-11:
How do I set a color for each key?  After selecting multicolor mode, I'm shown the following,

color1
color2
color3
and so on...

When I select color1, it gives me 3 options which are,

RGB Color1
Key range1
Delete

I'm not sure which option I need to select in order to choose a color.  Also, does color1 mean C?

onlaj, 2021-12-11:
You go yo Key range1 and set a range, for example:
Start: 50
End: 50
In RGB Color1 you set a color.
You do it for every key. I recommend using web interface for that.

clancel, 2021-12-11:
Web interface?  How do I get to the web interface?

onlaj, 2021-12-11:
https://github.com/onlaj/Piano-LED-Visualizer#web-interface

clancel, 2021-12-11:
Thank you for the link.  I connected to the web interface but I see blank pages with menu on the left sidebar only.

onlaj, 2021-12-11:
Can you make a screenshot?

clancel, 2021-12-11:
Hi.  Here's a screenshot.

192.168.178.108/24

clancel, 2021-12-11:
Hi.  Here's a screenshot.

https://paste.pics/FAV2K

onlaj, 2021-12-11:
This is really weird. Can you open development tools (F12), then "network" tab and then click on some page in the menu. Make a screenshot of both page and console.

clancel, 2021-12-11:
When I typed the address in the browser, nothing is showing.  Before, it at least showed the interface with menu only, but now it can't even connect.

clancel, 2021-12-11:
When I typed the address in the browser, nothing is showing. Before, it at least showed the interface with menu only, but now it's not even connecting or loading the page.

clancel, 2021-12-12:
Hi.  I am now able to access the menu contents in the web interface.  Here's a screenshot of the led settings,

https://paste.pics/633c826e847133ec6e8e18d5f4bf1fcb

I adjusted the colors directly on the pi visualizer menu yesterday, so the adjusted colors are shown here on the interface. It seems like the same note keeps changing to different colors when hit.  I would like the color to remain the same when hit again.  How can I achieve this?  Also, where is the key range1 option in the interface?

onlaj, 2021-12-12:
>I would like the color to remain the same when hit again. How can I achieve this?

Set range for all colors, so they don't interfere.

>Also, where is the key range1 option in the interface?

It's the slider under color selection.

closed 2022-11-02 by onlaj