onlaj/Piano-LED-Visualizer #295: Problem while loading OS Raspberry

Kir2406, 2021-12-12:
Hello everyone.
I built everything according to the instructions.
Installed the OS on microSD. But RPI is loaded sometimes. 
It can boot if you remove the screen, remove the memory card, and then collect everything and turning it on. In this case, it may start successfully 1 time. 1 time!
Then I turn it off via shutdown.
And the next time the diode blinks 4 times and that's it. Or the diode is constantly on. Or the diode does not light, but only a white screen works.

After that, I have to completely disassemble the RPI or reinstall the OS. 
And then, perhaps, the RPI will start.

I changed the OS, changed the microsd, started without a screen, etc.
Nothing works. No sequence of actions.

What is the problem? What am I doing wrong? Or could my RPI be broken?
I already regretted that I started to do it myself (((( just spent 120 bucks ((

onlaj, 2021-12-12:
Hello.
Does it happen when you install a PLV system image or just clean Raspberry OS?

It seems like a problem with RPi. Try to install a clean system, don't connect the led strip or screen, and check if the problem persists. Use SSH to check if Raspberry is working and to reboot.

Kir2406, 2021-12-12:
Yes, and PLV, and clean Raspberry from official website. The problem is the same.

I tried to run without screen, led strip. RPI only. and the problem is the same 😭
More precisely, it is, it is not. Unpredictable.

onlaj, 2021-12-12:
Where did you buy it? I would try to return it and get new one.

Kir2406, 2021-12-12:
Yes. I will try 😒