onlaj/Piano-LED-Visualizer #296: Not working with Raspberry Pi Zero 2

MiraiKuma, 2021-12-12:
Not working with the two.  Apparently it is because the rpi_ws281x doesn't have the zero 2 hardware revision in it yet. Not Putting this up here so others don't bombard you about it. 

Namespace(clear=False, display=None, port=None, webinterface=None)
Can't load input port: mio:mio MIDI 1 16:0
Can't load input port: mio:mio MIDI 1 16:0
Traceback (most recent call last):
  File "visualizer.py", line 82, in <module>
    ledstrip = LedStrip(usersettings, ledsettings)
  File "/home/Piano-LED-Visualizer/lib/ledstrip.py", line 35, in __init__
    self.strip.begin()
  File "/home/Piano-LED-Visualizer/neopixel.py", line 106, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, message))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)

MiraiKuma, 2021-12-19:
As an update, if you install manually the raspberry pi zero does work. So if you don't know how to do linux very well just wait till he posts a new update and you should get it to work on the raspberry pi zero 2. It works SO much better for me. The raspberry pi zero 1 was so sluggish I could barely get it to work and it was always cpu usage 80% for some reason. Installing manually and using the zero 2, it feels almost commercial grade.

MiraiKuma, 2021-12-19:
Nevermind. I spoke too soon. I twill work for about 30 seconds, then something happens and the cpu oges to 100% then goes back down to regular. This is exactly what was happening on my zero 1. I am not sure why I can't get this to work well.

onlaj, 2021-12-20:
I don't have RPi Zero 2 to test it, but I believe upgrading `rpi-ws281x` library will also help.
https://github.com/rpi-ws281x/rpi-ws281x-python/releases/tag/v4.3.1

MiraiKuma, 2021-12-22:
I tried upgrading the rpi-ws281x and that fixed the initial problem. Then I ran into a problem with the RTP-midi server so I updated that. Then I ran into some permissions problems. So I gave up and just installed from scratch which did the trick. I probably could have fixed it but installing manually was less than 10 minutes. Trying to narrow down all the errors was over 30 minutes lol.

AL1-1956, 2021-12-29:
Hello MiraiKuma an onlaj
First of all I'm a french "boy" of 66 years old , so my English is based on College Remembers ....

MiraiKuma ,I see that you have finally got the PianoVisualizer working on Pi Zero 2.
Can you please tell me what version of Raspberry you have installed (Buster or Bullseye) ??
I have a working PiZero 1 but I will receive soon a PiZero 2
I could try to put the working SDCard in the PiZero2 , but I understand that this did not wor for you

And a special thanks to onlaj for this GREAT project.
I managed to get it working without problem but for my own knowledge I waant a faster device
(I'm my own Santa)
For now I'm using midi over RTP but I will get pretty soon (I Hope) the Sevilla Gizmo 

Have a nice day and a Happy New Year
Alain

MiraiKuma, 2021-12-29:
AL1-1956,
Just follow the directions on the manual instructions on the front page. They will work for you on the Pi Zero 2. I installed Raspberry Pi OS Lite

Manual Instructions:
https://github.com/onlaj/Piano-LED-Visualizer/blob/master/Docs/manual_installation.md

On step #6 on the manual instructions replace
sudo wget https://github.com/davidmoreno/rtpmidid/releases/download/v20.07/rtpmidid_20.07_armhf.deb

with 

sudo wget https://github.com/davidmoreno/rtpmidid/releases/download/v21.11/rtpmidid_21.11_armhf.deb

AL1-1956, 2021-12-31:
Hello MiraiKuma and onlaj
Thanks for the answer
I have received the PiZero2W , but the header is not soldered to the PCB and I have absolutely "Zero" skills in soldering.
So I have ordered a hammering header (https://boutique.semageek.com/fr/1132-gpio-hammer-header-male-3004579660808.html).
Maybe this could help people like me who have 2 left hands
Happy End of 2021 to all

f84036980, 2022-01-12:
Thanks, @MiraiKuma,
I got my Raspberry Pi Zero 2 working now,  so much faster response!

AL1-1956, 2022-01-12:
Hello @f84036980 , happy for you
I have installed the PLV on my Pi Zero 2 but since it does not have , for now , a GPIO Header, I can only see via the Web Server that the CPU Charge is much lower than on my Pi Zero 1 .
So waiting to receive the "no need to solder" Header.

closed 2022-05-31 by onlaj