onlaj/Piano-LED-Visualizer #300: Software not opening on RPI.

Typro7, 2021-12-30:
Hello, I am trying my LED visualizer project, but when I put my SD card with the .img file it did not open up on the rpi.
The rpi has a screen on it and it has been only white ever since I connected it. I am using a 64gb SD card(bc it's the only card I have currently) and I formatted it to Fat32. Any help would be very appreciated. Thank you!

onlaj, 2021-12-30:
Hey.
You need to use software like Win32 Disk Imager or Etcher to mount image on sd card.

Typro7, 2021-12-30:
I will try this, thank you!

AL1-1956, 2021-12-31:
Hello onlaj , thank you very mutch for this wonderful job
If I may give some info about this problem.
I tried to use your img file with the official Raspberry Imager and it gives an error about the size of the image witch is not a multiple of 512.
So I tried with Win 32 Disk Imager and it did the job.

closed 2022-05-31 by onlaj