onlaj/Piano-LED-Visualizer #305: ImportError: No module ...

mostsu, 2022-01-05:
when i was starting my Raspberry pi and type this command
**"sudo -E python visualizer.py"**
it shows some error 
**"
import webcolors as wc
ImportError: No Mobule named webcolors
"**
I installed all required packages to python 3.9 version, But it seems like the system execute visualizer.py file by using interpreter python version 2.7.18

you can see the commands on images
`https://ibb.co/0QsWsvG`
`https://ibb.co/2vxj4Sh`

how to solve this problem ?

onlaj, 2022-01-05:
Hello.
This is correct command to start Visualizer:
`sudo python3 /home/Piano-LED-Visualizer/visualizer.py`

ishchawla, 2022-02-11:
hi! I have the same problem, even after using that command my visualizer is not starting. neither are any leds lighting up. When I enter that command, nothing happens. pls help
see these for reference:
[https://drive.google.com/file/d/1fyiKP462tv508VtBawZHUMJJ6hyFNBr_/view?usp=sharing](url)
[https://drive.google.com/file/d/1aC6PfVgtsPtFrV3nV3IHopA1brOtsZhx/view?usp=sharing](url)

onlaj, 2022-02-11:
@ishchawla Do you have crontab configured so Visualizer starts on boot? If yes then type
`sudo pkill python3`
it will stop any running python3 scripts.
Then start visualizer.py script and look for errors in output.

ishchawla, 2022-02-11:
Im not sure about that, I just used the preconfigured system image because I was having some issues while manually installing it. Should I still go on and type the pkill python3 command? 
Also, the visualizer.py script for starting is the one you mentioned above right?  This _sudo python3 /home/Piano-LED-Visualizer/visualizer.py_ one?

onlaj, 2022-02-11:
> Also, the visualizer.py script for starting is the one you mentioned above right? This sudo python3 /home/Piano-LED-Visualizer/visualizer.py one?

Yep, that's the one.

closed 2022-05-31 by onlaj