onlaj/Piano-LED-Visualizer #306: Not really an issue but some questions

AL1-1956, 2022-01-05:
Happy New Year to you onlaj and everybody reading
I'm french and I'm 66 years old , so my English is a little rusty.

I have a question about how to feed power to the LED ribbon.
For now I use the 5v and the ground from the pi to feed the LED ribbon and it's working fine with a 1m 144 LED.
Is it a good idea or is it better to use a separate power supply to feed the ribbon ?

My other question is this one
I use an M Audio Oxygen 88 as a keyboard and it has some knobs sliders buttons.
So it would be fantastic to have a "midi learn" function in PLV for mapping these to some of the options of the PLV , like stop play speed up speed down .... and maybe some of the LED settings
Do you think that this could be done ?

AnyWay , Thanks a LOT for this great job and have a nice day

Alain

onlaj, 2022-01-09:
Hello.

> Is it a good idea or is it better to use a separate power supply to feed the ribbon ?

I'm not an electrician so I can't really help you with that. I just followed instructions I found online.

> So it would be fantastic to have a "midi learn" function in PLV for mapping these to some of the options of the PLV , like stop play speed up speed down .... and maybe some of the LED settings
> Do you think that this could be done ?

That's a great idea. It would work nice not only for learning but also for changing other Visualizer's settings. My piano doesn't have such features, but if some more advanced keyboards can send custom messages through midi I could make them work as "hotkeys". Adding it to my to-do list.

AL1-1956, 2022-01-09:
Hello onlaj.
Thank you very much for your answer
About the "midi learn" function , it works like this
You have a device (midi keyboard without sound) which sends a midi message to another device. (desktop synth module)
Each message has it's own role (ie: CC = 7 is the global volume , another CC sends pan ,another sends reverb level ....)
So you can control the sound of the device witch listens to the master
But you can also send other types of messages wich are real time messages (start stop the sequencer)
it's up to the receiver to use (or not ) what it gets from the master keyboard.
The list of midi messages is well defined but the receiver is free to do what it wants with what it gets.
So for example , the master could send CC23 and the receiver could use this to change the color of the LEDS
What makes the force of midi learn is that you choose what function you want to activate on the receiver and by moving the knob , slider , button on the master the learning is done.

SirLefti, 2022-01-10:
Hello Alain
> Is it a good idea or is it better to use a separate power supply to feed the ribbon ?

I actually wired it the same way without any issues, running it for a few month now. It is definitively better to use a separate and more powerful power supply from a specification and safety view. However, if you restrict the brightness to 10% or so, playing on it should not be too much of an issue, because your ten fingers should not be able to press too many keys at the same time :). I am also using the animations on a low brightness. Other colors than white should also reduce the power draw per LED at the same brightness.

As far as I know the GND and 5V Pins are directly connected to the USB power input. This means, that those pins are mainly restricted by the maximum load the power supply can handle. Unlike the data pins, which are connected to the CPU and thus should not exceed 16 mA per pin.
You can potentially also go the other way around and feed the PI via those pins with an external power supply. 

I do not recommend doing this, because it was not designed this way and PCB revisions may vary. It might work very well, but it is your own risk.

AL1-1956, 2022-01-10:
Hello SirLefti
Thank you for your answer.
I'm not a Wizard like Jordan Rudess (https://www.youtube.com/watch?v=NI_H16zIJfI&ab_channel=PedroAgraSabino)
, so I think that the option of draining power for the LED ribbon from the Pi is safe , and I will definitively take in consideration your advice about reducing the glowing of LEDs.
I'm wating to receive a 2m LED ribbon from AliExpress , because the ones I got from Amazon are only 1m long and I'm not an expert with soldering .
The keyboard I'm using is a 88 notes so 1m is not enough to cover all the keyboard.
Anyway thank you for your advice and have a nice 2022.

Alain

SirLefti, 2022-01-15:
> You can potentially also go the other way around and feed the PI via those pins with an external power supply.

I just wanted to add something here, in case you or someone else reads this and tries to implement it like that.
By connecting power to 5V and GND pins you are bypassing the fuse, which might damage the board, especially the CPU. Take extra care here.

AL1-1956, 2022-01-16:
Hello SirLefti
Thank you for this info.
I feed the Pi via it's mini USB port either with USB port of my computer or (if I want to be "computer free") , with an external power supply.
And I use pins 2(5V)  6(GND) and 12(GPIO 18) to feed and control the LEDs ribbon with a bypass GPIO header.
[image]
Anyway Thaks for the advice and have a nice day

Alain

closed 2022-05-31 by onlaj