onlaj/Piano-LED-Visualizer #312: LED Lighted up but ...

mostsu, 2022-01-08:
Hello Onlaj.
I use a Raspberry pi zero 2W. 
Now my raspberry pi does it work and detect my piano (CASIO PX-130) you can see the OS show massages (see in a pic)
https://ibb.co/0XZH1WN
and freezing at last message "Starting Webinterface"
I try to press some piano keys and then i can see LED light up on WS8212B as white color but the 1.44inch LCD hat always white screen so i can't control menus for select other colors (see in a clip vdo)
https://vimeo.com/user162902838/review/663785130/63d4a7445c
https://ibb.co/bz5q39V
thanks for help. :)

onlaj, 2022-01-09:
Hello.
Can you connect to webinterface? You can control your settings from there.
As for screen, try to run /tests/test_screen.py
If it doesn't show anything on your screen, try to unplug strip's wires (as well as your extension board) from your RPi and run script again.

mostsu, 2022-01-12:
Hello Onlaj, Thank for answer me.
I don't know how to open the webinterface, how to open it ?
I tried to execute 'test_screen.py' file and it works in case the LCD hat connect to Raspberry pi board directly :).
https://vimeo.com/665096225
______________
But why when i use LCD hat connect to expansion board and take the expansion board connect to Raspberry pi board it does not work :( ?
you can see in this vdo.
https://vimeo.com/665096531
______________
Thank for help :)