onlaj/Piano-LED-Visualizer #314: 3D Printer Service Issue

AL1-1956, 2022-01-10:
Hello onlaj , sorry to bother you with that but.

I tried to have the case for the PLV made by a 3D Printer Service but they want STL files with separated pieces

So I tried to use FreeCad to separate the pieces of your STL file but I cannot figure how to do that.

Any kind of advice will be great.

Have a nice day (or night , I don't know )

Alain

onlaj, 2022-01-12:
I searched my hard drive and found already splitted files, here you go

[case_multiple_parts.zip](https://github.com/onlaj/Piano-LED-Visualizer/files/7854170/case_multiple_parts.zip)

AL1-1956, 2022-01-12:
Great !!!
I finally found a way to do the job thanks to an online help
https://jugulaire.github.io/Freecad-decoupe-piece/
it's in French but you can use Google Translate to have it in your language.
But since it's a "workaround" i better use your original parts
Have a nice (day , night , afternoon , morning ....)
Alain

AL1-1956, 2022-01-12:
Hello again
I opened your files in zip archive and I have a question.
Are the three buttons in one piece or in two pieces (the button and the support) ?
So if i ask to my 3D Printer Service to do the job for every piece I will have
case_down
case_up
round button
first button push
first button support
second button push
second button support
third button push
third button support
=> 9 separate pieces or only 6
Thank You
Alain

onlaj, 2022-01-12:
You can send them all files and tell to print:
top
bottom
3 small buttons
1 round button
They can figure out which files are best to use.

AL1-1956, 2022-01-12:
hello onlaj once again.
I'm sorry to insist but maybe my question was not clear enough (or maybe my English is to rusty...)
Can you tell me if each of the three push buttons is made of one piece or two pieces because when I burst your model in FreeCAD it gives me two pieces for each push button , the button itself and what seems to be a support
[image]
[image]
Sorry again for bothering you with my (very old) newbee questions.

onlaj, 2022-01-12:
Oh yeah, sorry, I misunderstood you. This is one piece.

AL1-1956, 2022-01-13:
Ok Thanks a lot and have a nice day 

Alain

closed 2022-05-31 by onlaj