onlaj/Piano-LED-Visualizer #317: LED not lighten up.

mostsu, 2022-01-12:
Hello Onlaj.
The software of Raspberry pi zero can booting up and my piano being detected.
But when I pressed some keys on the piano, LED not lighten up :(
you can see my physical set up in this pic
https://ibb.co/0mkrwTh
is this set up correctly ?
___________________________________
I observed the WS2812B LED strip that lighted up some point at first and last when I plugged an AC/DC adaptor, is this normally ?
Previously the LED strip it works but the first point of LED strip not show white light.
you can see in the video
https://vimeo.com/665099980
____________________________________
Thank for help.

onlaj, 2022-01-12:
Hello.
What about led animations, do they work?

mostsu, 2022-01-12:
Not work.
I tried to animation LED is nothing happened.

onlaj, 2022-01-12:
I can only suggest to double check your wiring. It seems ok on picture, but maybe it doesn't connect somewhere.

closed 2022-05-31 by onlaj