onlaj/Piano-LED-Visualizer #330: Modification of tempo with web interface

theblackhat17, 2022-01-24:
### Discussed in https://github.com/onlaj/Piano-LED-Visualizer/discussions/323

<div type='discussions-op-text'>

<sup>Originally posted by **theblackhat17** January 18, 2022</sup>
Hey, I have just configured the entire piano project, everything works perfectly I am delighted!
Today, I would like to add a functionality, it is to modify the tempo of a piano piece using the web interface directly. Indeed, we can do this from the small OLED screen, so I would like to know if I can retrieve the variable from the python program to call it in the html page and let the user modify his piano tempo to play more or less fast. 
I will add the speed modes in the web interface directly with the attributes : slow(25%) / medium(50%) / fast(75%) / hard(100%). All this by changing the tempo percentage which is 100% at the beginning.
I saw that there was the function "**set_tempo**" or "**song_tempo**" present in the file "_learnmidi.py_".

`self.set_tempo = 100` or `self.song_tempo = 500000` or `self.ticks_per_beat = 240` 

In my new page named "songspead.html" and added in the menu, I added the four options in the list like this :                      `<option value="0">Slow</option> <option value="0">Medium</option> <option value="0">Fast</option> <option value="0">Hard</option> `

Do you think it would be possible?
Thanks for your help!

Th3_bl@ck_h@t</div>

onlaj, 2022-01-24:
Hello.
What about instead of explaining how to do it, I just add this feature? :D

theblackhat17, 2022-01-25:
> Hello.
> What about instead of explaining how to do it, I just add this feature? :D

Yes If you want, you can add this feature ;) If you want you can explain me how to do this. Thanks a lot

onlaj, 2022-01-27:
Done.
[image]

If you notice any bugs report them in this thread.

theblackhat17, 2022-01-27:
> Done. [image]
> 
> If you notice any bugs report them in this thread.

You are the `BOSS`!! Thanks a lot for your Work **🙏** You help me very much

theblackhat17, 2022-01-27:
I will try it tomorrow, 
By having created this page, you are unblocking me from a problem. Thank you again for your help.

closed 2022-05-31 by onlaj