onlaj/Piano-LED-Visualizer #334: Learn midi repeat feature.

Pausejo, 2022-02-10:
Hey Aleksander, I was wondering if it would be possible to put a repeat feature in the learn midi section.

I'm currently trying to learn some "hard" midis sections and I find the mechanisms to learn a bit difficult tricky for finding the exact place which I want to learn, and when I finish the section to push the start button again.

For this, I would like to add a repeating function, but instead percentages, just put from measure number to end measure number.

[image]

I'm not sure if it's possible, but thanks to read me anyway!

onlaj, 2022-02-25:
Hey,

I tried what I could, but I find it very difficult to set a range based on measures, especially on midi files with an inconstant tempo.

I wanted to make it easier to find the desired range of points. At first, I tried to display sheet music using this [library,]() but it works only with xml or abc files, and it seems that converting midi to those formats is a little bit tricky.

So for now, I added small flag icons with an explanation popup.
[image]

You can press it to set the current track time as the start/end point. It's not ideal because it still uses integer percentages as points. I will try to make it more precise in future updates.

Pausejo, 2022-02-25:
Good news to hear that, I'll take a look today. Thanks for your great effort!
I think it will be much easier to practice the desired fragment over and over again :)

closed 2022-05-31 by onlaj