onlaj/Piano-LED-Visualizer #337: LED Colors are not right

Ferdinandehni, 2022-03-04:
Hello,

Thanks for the Piano LED Visualizer. It is great but I have an issue with colors.

I set the backlight to 30% red, and the LED Strip mixed up the colors. I had the same Issue, when I run the example code from Adafruit ([Link](https://learn.adafruit.com/neopixels-on-raspberry-pi/python-usage))  I changed `ORDER = neopixel.GRB` to `ORDER = neopixel.RGBW` and I uncommend/commend the corresponding parts in the end and it worked.

How can I change this in the Visualizer to solve the problem?

[image]
[image]

onlaj, 2022-03-04:
Hello.
What is the name of this led strip? SK6812?

Ferdinandehni, 2022-03-04:
Oh no. You are right, it is an SK6812RGBW. I bought the parts not carefully enough.

closed 2022-05-31 by onlaj