onlaj/Piano-LED-Visualizer #345: Visualizer not Working

psscin01, 2022-03-25:
Hi Onlaj,

I have made this project before and it works wonderfully! I tried making it again and I was unable to connect to the visualizer. I am using a Pi zero 2 btw. When I give the command, I receive this error:
[image]
Any thoughts as to what the problem might be? As always, thanks for everything!

psscin01, 2022-03-25:
I mean that the visualizer will not run unless I use the command. In other words, I gues the  crontab isn't being recognized at boot.

onlaj, 2022-03-25:
And what does it look like?

`@reboot sudo python3 /home/Piano-LED-Visualizer/visualizer.py &`

^ this is the right one

If launching script using command from your screenshot works fine then it should also works from crontab. You can try to reinstall cron just to be sure.

psscin01, 2022-03-25:
Yes, that was the one I have in there. I'm not really sure why it only works from the command line, but I will try reinstalling cron. I apologize for my lack of knowledge, but what would the command be for that?

Another question: Would having another visualizer on simultaneously on the same network effect this one?

onlaj, 2022-03-25:
`sudo apt-get install --reinstall cron`

> Would having another visualizer on simultaneously on the same network effect this one?

Yes, it will probably fail at launch because port will be already taken. There is launch command that allows you to change default port for web interface. Or you can disable web interface at all on one of your visualizers.

psscin01, 2022-03-25:
Ah that must be it then. I am trying to run multiple visualizers in a classroom, so that's probably where the problem is coming from. Would I need to find an open port using the command 'netstat -lnu | grep udp | awk '{ print ( $4 ) }' | awk 'BEGIN{FS=":"} { print $(NF) }' | sort -n | uniq'? Then would I direct to that port in the crontab file?

onlaj, 2022-03-25:
Yep, the command looks like this: 

`sudo python3 /home/Piano-LED-Visualizer/visualizer.py --port 5000`

I'm not sure how it will work with recently introduced websockets, but those are used only for synchronising sheet music with learning, so it shouldn't be a problem.

psscin01, 2022-03-30:
So I tried to adjust the port in the crontab, but it still wouldn't work. I ran `sudo nano /etc/profile` and entered `sudo python3 /home/Piano-LED-Visualizer/visualizer.py` at the bottom of the script there. It worked!

closed 2022-03-30 by psscin01