onlaj/Piano-LED-Visualizer #348: Errors running the program

CathesisGM, 2022-04-03:
I manually installed the software to my pi 3B, and when I try and run visualizer.py from the terminal I'm given this error:
[image]

I'm a newbie when it comes to this kind of stuff so some help would be great :)

onlaj, 2022-04-03:
Hey, run visualizer with "sudo" permissions
`sudo python3 visualizer.py`

CathesisGM, 2022-04-03:
> Hey, run visualizer with "sudo" permissions `sudo python3 visualizer.py`

Thanks, that worked. visualizer.py seems to run now, but my LED strip just goes bananas and nothing I do on the web interface/piano seems to do anything to it? I'm not sure if I'm missing something obvious again or not.

onlaj, 2022-04-03:
> goes bananas

I'm not sure what you mean by that, but check your wirings.

CathesisGM, 2022-04-03:
Apparently my ground wire fell out haha. Works great now, thanks a ton.