onlaj/Piano-LED-Visualizer #349: PLV 1.3 no autoload

AL1-1956, 2022-04-21:
Hello Aleksander
I downloaded and installed the new version of PLV on my Pi Zero W and I have this issue

The PLV 1.3 does not start when I boot the Pi , however it starts when I type 

cd /home/Piano-LED-Visualizer
"sudo python3 visualyzer.py 

I checked the crontab and it is OK , so I am a little stuck with that problem

Can you confirm that it MUST autoboot with the image file you give to us

Thank You and have a nice day

Alain

AL1-1956, 2022-04-22:
Hello again.
I tried to log the crontab by using this 
@reboot sudo /usr/bin/python3 /home/Piano-LED-Visualizer/visualizer.py >> /home/log

and the log file is empty , so I guess that the cron is not working .
It worked perfectly with PLV 1.2 so I'm going back to PLV 1.2 unless somone can confirm that PLV 1.3 is working with the img file and without any modification....

onlaj, 2022-04-22:
> I'm going back to PLV 1.2 unless somone can confirm that PLV 1.3 is working with the img file and without any modification....

Hello, After creating an img file for release, I always write it on a new sd card to test if everything is OK. I'm using that SD card now and cron is working fine. It *should* work without any modifications, so I'm not sure why it's not the case for you. Maybe try to reinstall Cron?

AL1-1956, 2022-04-22:
Hello , thank you for your quick answer.
I moved back to PLV 1.2 wich works as is with the img file.
I have at the moment only one SD Card so I cannot make a check on the cron feature between PLV 1.2 and PLV 1.3.
May be you know a way (other than the log file) to check if the crontab has been read .
Anyway , have a nice day

Alain

AL1-1956, 2022-04-22:
OK , work in progress.
I am back with PLV 1.3 and I have only this single instruction in the crontab

"@reboot date > /home/pi/log"
and after reboot
pi@raspberrypi:~ $ cat log
Fri 22 Apr 21:05:30 BST 2022
so the cron is working but for some unknown reason it does not start the new PLV 1.3

Alain

onlaj, 2022-04-22:
How exactly does your crontab command look like?

AL1-1956, 2022-04-22:
ok so now I have both PLV1.2 and PLV1.3 installed on the same SD cards which is the 1.3 image
my crontab is
"@reboot sudo python3 /home/PLV_1.3/visualizer.py &" => this one does not work
if i replace by
"@reboot sudo python3 /home/PLV_1.2/visualizer.py &" => this one works

but both work if instead of using the crotab i type
sudo python3 /home/PLV_1.3/visualizer.py &
or
sudo python3 /home/PLV_1.2/visualizer.py &

onlaj, 2022-04-22:
Why the directories under /home are called "PLV_1.2" and "PLV_1.3"? I'm pretty sure I called them Piano-LED-Visualizer

AL1-1956, 2022-04-22:
I rename  so i can know which is 1.2 and wich is 1.3

AL1-1956, 2022-04-22:
Great News (for me at least...)
I removed the X attribute of the visualizer.py file with "sudo chmod -X visualizer.py" and now PLV 1.3 is  working with the crontab
Mistery .....

AL1-1956, 2022-04-23:
It's no more working so I guess the +X attribute was not the solution.
Anyway I have now a working PLV1.2 and a "not very well" working PLV1.3.
So I can use the app while I make more research
I will no more bother you until I make real progress 
The PLV1.2 IS working and i am very thankful to you for this great app

Alain

AL1-1956, 2022-04-24:
Ok for Now i give up with the cron method and use the profile method suggested by Roloo38 in closed issue #153 
And i'm happy to thank him because it works every time (tried 5 times in a row) 
Simply add this line at the end of  /etc/profile
"sudo -E python3 /home/pi/PLV1.3/visualizer.py &"

I do not know why it's not working with the cron but for  now I don't care
Have a nice day
Alain

AL1-1956, 2022-04-26:
Hello .
I made real progress.
The crontab works fine with something like
"55 10 26 4 3 sudo python3 /home/Piano-LED-Visualizer/visualizer.py &"
but does not work with
"@reboot sudo python3 /home/Piano-LED-Visualizer/visualizer.py &"
So I guess that some resource is not available at boot and maybe I need to add a "wait something" somewhere ...
I am going to compare code for PLV 1.2 with code for PLV 1.3 , but my knowledge of Python is very (very very) basic.

onlaj, 2022-04-26:
Maybe try something like
`@reboot sleep 60 && sudo python3 /home/Piano-LED-Visualizer/visualizer.py &`

AL1-1956, 2022-04-26:
Hello , this works !!!
Thank You very much
:thumbsup: :thumbsup: :thumbsup: :thumbsup: :thumbsup:

onlaj, 2022-04-26:
I'm glad to hear that.
I need to get some HDMI display. I usually control my RPi through SSH, so I can't test how the script behaves without wi-fi.
I suppose it takes longer for your RPi to get a wifi connection than on my setup.

onlaj, 2022-05-11:
@AL1-1956 I tried two different HDMI cables, two monitors and nothing works. I think my HDMI output is damaged and I can't find any Zeros in reasonable price.

Can you check something if you have some spare time?
Type `sudo raspi-config` and go to System Options -> Network at boot -> set as Enabled
Then remove `sleep 60 &&` part from crontab.
This change will make RPi wait for network connection before finishing boot sequence.

AL1-1956, 2022-05-11:
Hello
I tried the trick , but this does not work.
Is there a way to check if the "Network at boot" is really enabled (other than the raspi-config) ??

onlaj, 2022-05-11:
I don't think so.
I blocked all ports (except SSH's) and started Visualizer without any problems, so I think lack of network is not an issue for a script. It was also confirmed by your empty log file because it means that script didn't even tried to start.
On the other hand, you said that version 1.2 works fine, and that implies cron is working fine and it's the script's update fault. I'm really confused.

AL1-1956, 2022-05-11:
If there is some kind of log or something that can be useful for you , let me know an I will send to you
I'm out for a few hours to have a ride with my dog . Thank for your effort

Alain

AL1-1956, 2022-05-11:
Ok I am back and I am going to try this
As I have now 2 16Gb SD Cards , I will write one with the PLV1.2 image and the other with the PLV1.3 image 
(out of the box except for the wpa_supplicant.conf and the ssh files that I need to connect on Wifi to my PI)
The Pi itself is a Pi Zero WH (wifi with header) and I only plug it to my PC (Intel) with the USB power socket.
So I access the Pi in Wifi via (dos console and SSH) .
To be continued

AL1-1956, 2022-05-11:
So The first SD Card with PLV1.2 autoboots  and the second one with PLV1.3 does'not.
I was thinking (maybe completely wrong ) of some kind of DHCP conflict since the IP adress is the same in both cases.
But I do not know how to change it on linux at next reboot without cutting my leg.

Buleste, 2023-06-08:
Did you ever solve this issue? 1.2 autoboots fine. 1.3 and 1.4 will only autoboot if WiFi is set up which is an issue as many places where my friend gigs do not have WiFi.

onlaj, 2023-06-09:
@Buleste @AL1-1956 Correct, for now you need wifi. I'm still waiting for Raspberry Pi availability, once I get one I will release new version.

Buleste, 2023-06-09:
Thanks for the reply. I'll go back to 1.2 then.

onlaj, 2023-09-17:
I released version 1.5, if you have any questions feel free to reopen this issue.
https://github.com/onlaj/Piano-LED-Visualizer/releases/tag/v1.5

closed 2023-09-17 by onlaj