onlaj/Piano-LED-Visualizer #352: Midi with USB (more a question than an issue)

AL1-1956, 2022-05-01:
Hello
There are 2 USB ports on the Pi Zero 
One is used to feed the power to the Pi and the other is used to connect mouse or keyboard

Is there a way to send and receive MIDI informations on any of these USB ports ?? 
Thank You for your answer

Alain

onlaj, 2022-05-01:
Hello.
As long as on the other side there is a midi device you can transfer midi informations using the second usb port.
If you want to send midi between two non-midi devices (for example Raspberry and PC) you can use this device: [MIDI USB-USB](https://compasflamenco.com/midi-c-3/midi-usbusb-p-4.html)

You can also configure Raspberry in [OTG Mode](https://gist.github.com/gbaman/50b6cca61dd1c3f88f41) but it won't work through usb hub, so you can have only one usb device connected.

AL1-1956, 2022-05-01:
Thank You for the quick answer .
There is something , I do not understand.
Th Pi can be fed with either of the 2 USB ports .
When I plug the cable from the Pi into one of my PC USB port the LCD sreen lights up with both of them. 
What is the port you call the second one , is it the one close to the HDMI or is it he other one ??

And when I go into the list of available midi ports on the Pi , I do not see any one them .
I see only network ports (RTP or Wifi) .
Is there something I am missing or do I need to modify the connect_all script ??
Thank You

Alain

onlaj, 2022-05-01:
> What is the port you call the second one , is it the one close to the HDMI or is it he other one ??

The one closer to HDMI, yes.

> And when I go into the list of available midi ports on the Pi , I do not see any one them .

What do you mean by that? Where do you look for midi ports and what did you connect to your RPi?

AL1-1956, 2022-05-02:
Okay , I need to explain my configuration.
I have one midi keyboard (Oxygen88) which can send midi via midi OUT and via USB
I have one midi desktop unit (BOSS DS-330) which has midi IN - OUT and THRU
I have one PC which has 2 free USB ports
I have one Sevilla USB-USB and one Sevilla UHM2
For now  
The Oxygen  can send MIDI to  the BOSS and can send MIDI to the PC (for sequencing in Musescore )
The PC can send MIDI to the BOSS (for playing from Musescore or Van Basco Karaoke)
(The UHM2 merges MIDI from the Oxygen and from the PC thru the USB-USB and sends it to the BOSS)
And it's working great.

I was thinking of using the MIDI THRU of the BOSS to send all incoming MIDI (OXYGEN and PC) to the Pi
so that the LED ribbon lights UP when it receives MIDI from the OXYGEN (when I play live) and from the PC (when Musescore sends MIDI) . I do not need the MIDI play and the MIDI learn functions of the Pi only the lighting

Is it possible ??
Than You

Alain

[image]

AL1-1956, 2022-05-02:
Work In Progress 😺 
I connected my Midi Mate (USB - Midi) interface to the PI and the ports have been updated with 
Input XP Midi Mate 1 20:0
Playback XP Midi Mate 1 20:0
So I guess , i am on the right way.

onlaj, 2022-05-02:
Your setup seems to be far more advanced than mine, I'm afraid that I can't be much of a help. Nevertheless, good luck :)

closed 2022-11-02 by onlaj