onlaj/Piano-LED-Visualizer #354: New Update Visualizer doesn't work

justkatia, 2022-05-04:
Since I've updated the new version the Visualizer menu doesn't appear on my RPi screen anymore. I tried to look for solutions in the previous issues but none of them worked for me. I removed and reinstalled the Piano-LED-Visualizer but this doesn't work either. I would be thankful for your help :)

onlaj, 2022-05-04:
Hello.
Can you connect to your RPi through SSH and manually start Visualizer?

Use those commands:
`sudo pkill python3` to stop working python script

`sudo python3 /home/Piano-LED-Visualizer/visualizer.py` to start script

If it shows any errors copy and paste them here.

justkatia, 2022-05-04:
Yes ssh does work.  I tried and I have an import error: ImportError: cannot import name 'safe_join' from 'werkzeug.utils' (/usr/lib/python3/dist-packages/werkzeug/utils.py)

[image]

onlaj, 2022-05-04:
Hmm, it should be installed by default, but try
`sudo pip3 install Werkzeug`

justkatia, 2022-05-04:
It says: Requirement already satisfied: Werkzeug in /usr/lib/python3/dist-packages (1.0.1)
[image]

onlaj, 2022-05-04:
Ok, it seems you have older version installed, try
`sudo pip3 install Werkzeug --upgrade`

justkatia, 2022-05-04:
thanks. apparently I could install a new version but now I get another import error if I run the commands xD
[image]

onlaj, 2022-05-04:
Maybe try to reinstall all required packages?
`cd /home/Piano-LED-Visualizer`
`sudo pip3 install -r requirements.txt --upgrade --force-reinstall`

justkatia, 2022-05-04:
wow thank you :) it worked !

BaseD75, 2022-07-03:
Great thank you, had the same issue

vulcanjedi, 2023-06-25:
Thank you! Me too!