onlaj/Piano-LED-Visualizer #355: No auto start, unless ssh?

Logan-Orr, 2022-05-07:
Hello! In the new PlV 1.3 update, the autostart is no longer working, i know how to manually start it and it works, but when i put the sudo python3 /home/Piano-LED-Visualizer/visualizer.py &, it does not start, but when i ssh into it, it automatically starts up and the menu is shown with no commands entered, and if i close the ssh window, it stops working, how can i fix this?

onlaj, 2022-05-07:
Hello.
https://github.com/onlaj/Piano-LED-Visualizer/issues/349#issuecomment-1109641040

I will try to fix that in future updates, for now try this workaround.

Logan-Orr, 2022-05-07:
So I looked at that yes, where would i find the crontab?

onlaj, 2022-05-07:
Type:
`sudo crontab -e`

Logan-Orr, 2022-05-07:
Thanks bro u respond so quickly

Logan-Orr, 2022-05-07:
Also, this is a weird error, the visuizer is working, but i can not enter any commands because of this infinite loop of error
[image]

onlaj, 2022-05-07:
Just press CTRL + C

Logan-Orr, 2022-05-07:
Ok

Logan-Orr, 2022-05-07:
Socket listening on: 192.168.1.185:8765
Socket listening on: 192.168.1.185:8765
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
Socket listening on: 192.168.1.185:8765
Socket listening on: 192.168.1.185:8765
Socket listening on: 192.168.1.185:8765
Socket listening on: 192.168.1.185:8765
Socket listening on: 192.168.1.185:8765
Socket listening on: 192.168.1.185:8765
Socket listening on: 192.168.1.185:8765
Socket listening on: 192.168.1.185:8765
Socket listening on: 192.168.1.185:8765
Socket listening on: 192.168.1.185:8765
Socket listening on: 192.168.1.185:8765
Socket listening on: 192.168.1.185:8765
Traceback (most recent call last):
  File "/home/Piano-LED-Visualizer/visualizer.py", line 17, in <module>
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    from webinterface import webinterface
  File "/home/Piano-LED-Visualizer/webinterface/__init__.py", line 43, in <module>
    from webinterface import views
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
  File "<frozen importlib._bootstrap>", line 982, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 925, in _find_spec
  File "<frozen importlib._bootstrap_external>", line 1349, in find_spec
    self._target(*self._args, **self._kwargs)
  File "/home/Piano-LED-Visualizer/webinterface/__init__.py", line 36, in start_server
  File "<frozen importlib._bootstrap_external>", line 1321, in _get_spec
  File "<frozen importlib._bootstrap_external>", line 1450, in find_spec
    asyncio.run(main())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
  File "<frozen importlib._bootstrap_external>", line 1490, in _fill_cache

Logan-Orr, 2022-05-07:
Am i missing a requirement?

onlaj, 2022-05-07:
Are you trying to run script manually now or are those logs from the start-on-boot script?

Logan-Orr, 2022-05-07:
Start on boot

onlaj, 2022-05-07:
Then don't worry about it. If it's the same case as in the other issue, then it's an error caused by the script trying to run before getting a wifi connection.

It can be temporarily fixed by adding a delay to the script's start.

Logan-Orr, 2022-05-07:
Ohhh, it could possibly be from the sudo python3 /home/piano etc. start script that i left in profile

Logan-Orr, 2022-05-07:
Yea thats what it was

Logan-Orr, 2022-05-07:
I have a idea, i was using a synthizer that was for raspi and it was headless, it would be awesome if you could combine the synth with the plv, and have a soundfont menu, the installation was one command and was fast, i can send you a link

fredpineau, 2022-05-29:
Hello, I have created a specific systemctl service in /usr/lib/systemd/system like this one and it works fine on my side: 

**pi@raspberrypi:/usr/lib/systemd/system $ cat piano-visualiser.service**
[Unit]
Description=Piano Visualizer
After=network-online.target
Wants=network-online.target

[Install]
WantedBy=multi-user.target

[Service]
ExecStart=sudo python3 /home/pi/Piano-LED-Visualizer/visualizer.py --display 1in3
Type=simple
User=pi
Group=pi
WorkingDirectory=/home/pi

and then enable it :

`systemctl daemon-reload`
`systemctl enable piano-visualiser.service`

closed 2022-11-02 by onlaj