onlaj/Piano-LED-Visualizer #357: No different colors for left and right hand while playing with Syntesia

Sonar10, 2022-05-19:
Hello,
I have a problem, when I try to learn to play with Synthesia, my LEDs don't indicate the color of corresponding hand (blue, green), all LEDs light in white. I tried different configuration but without success. I have connections like this: RPi->USB OTG-> Piano and RPi->USB OTG->Sevilla's Soft MIDI USB-USB->PC

Could you help me?

onlaj, 2022-05-19:
Hello.
Did you change your Synthesia's settings to "Finger-based channel"? It's under MIDI USB-USB output -> key lights.

Sonar10, 2022-05-19:
Yes, I did: Music output ->MIDI USB-USB->Key Lights ->Finger-based channel.
I'm sending pictures, because LEDs have weird behaviour.
[image]
[image]
[image]

onlaj, 2022-05-19:
What about other output settings in Synthesia? Set "my notes" as disabled, same with "prevent local notes".
If it doesn't help, there is a workaround. Go into "Led settings" -> "Skipped notes" and select "normal".

Sonar10, 2022-05-19:
I did all things that you mentioned above - now LEDs not showing the first notes at all, but if I press them and the next notes are the same they light up properly in blue and green. I'm sending more pics:
[image]
[image]
[image]
[image]
[image]
[image]

onlaj, 2022-05-19:
Did you try other midi files? Does the same thing happen with Synthesia's default songs?

Sonar10, 2022-05-19:
I think you were right. I downloaded some songs and it works good. What I've noticed that not all MIDI files show hands - I assume that the MIDI file must be prepared somehow by creator to indicate hands right? And the second question do you know any good MIDI files repository?

onlaj, 2022-05-19:
[Musescore](https://musescore.com/) with some browser extension for downloading files.

closed 2022-05-31 by onlaj