onlaj/Piano-LED-Visualizer #360: LED strap operation is not possible.

tdoypwls, 2022-06-08:
I followed the guide exactly and confirmed that the LCD booted normally.
However, the LED strap only blinks and cannot be control at all.

All operations such as MIDI playback, piano playing, and LED color change are impossible.

The LED strap I bought is 144leds, IP67, 2m.
I connected the LED strap wire to Gnd(-) and Gpio15(control) of Raspberry Pi PCB correctly.

Help plz...

LED strap link : opt(144leds, IP67, 2m.) // 
https://a.aliexpress.com/_mOZELuA

[image]
[image]

https://user-images.githubusercontent.com/107099691/172548997-041fcc67-02a6-4af2-85d0-bd7f4dc7b7b5.mp4

onlaj, 2022-06-08:
Hello.
It's really hard to tell from the photos you posted, but are you sure the wiring is correct? It seems like data cable is going to the wrong hole in the socket.

tdoypwls, 2022-06-08:
Yes. It didn't work, so I tested all sockets. I also tried the opposite.

as in the video
Connected correctly to GND and GPIO 15 of Raspberry Pi.

GND is connected to the 3rd port from the left, and the LED data GPIO is connected to the 6th port from the left.

The LED does not work even if I reverse it to the right just in case.

onlaj, 2022-06-08:
I meant the other side of cables. Does it go to the correct wires on led strip? From the pictures it looks like data (from RPI) is connected to (+) (from the led strip).

tdoypwls, 2022-06-08:
Yes, there are 3 cables attached to the LED strip.

The order is + / data / - .

+ is the red cable

data is a black cable

- is the white cable.

- is connected to GND of Raspberry Pi board and - of 5v,

+ is only connected to 5v +.

data is connected to GPIO 15 with black and yellow wires.

tdoypwls, 2022-06-08:
The LED strap I have is

The order is different from the last photo example image.

my strap is
(+) / data (-) and

example image is
(data) blue
(+) red
(-) It is in black

onlaj, 2022-06-08:
> The order is different from the last photo example image.

Yeah, sorry, it was little missleading. I checked my led strip and data is also in the middle.

Just to make sure it's not a problem with my code you can try to run [strandtest.py](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/tests/strandtest.py) . If it still doesn't work I suspect there is something wrong with either RPI gpio pins or led strip.

tdoypwls, 2022-06-08:
Thank you very much.

Let's try it again.

If it doesn't work, I guess I'll have to buy the strip again...

tdoypwls, 2022-06-08:
I ran strandtest.py. However, No. The module named rpi_ws281x is printed.
I did the following:

cd ~
sudo pip uninstall rpi_ws281x
sudo pip install rpi_ws281x
sudo pip3 uninstall rpi_ws281x
sudo pip3 install rpi_ws281x

But it keeps saying that the module does not exist, what should I do
[image]
?

tdoypwls, 2022-06-08:
Trying strandtest.py doesn't change anything. I think you should buy a strap controller and check if it's a strap problem. thank you

closed 2022-11-02 by onlaj